RobotLocomotion/drake #18682: [contact_manager] Recursive-ness in discrete updates?

EricCousineau-TRI, 2023-01-28:
### What happened?

I am writing controllers that rely on zero-order holds because it can make things a nicer place in the real world.
Per https://github.com/RobotLocomotion/drake/pull/18551#issuecomment-1384319905, I am using multiple initializations to arrive at "steady state" for discrete state.

I naively expected that if I saw the exact same `(q, v, tau)` for a plant, then I wouldn't have to worry about any other state.

However, it seems like there is extra state, possibly in the contact manager, that will "accumulate" across my calls.

I've added the example to #18235, in 9a475540c5:
https://github.com/RobotLocomotion/drake/blob/9a475540c5d844c2b5b2d5b37eca785f0f758882/examples/instability/test/plant_hidden_state_test.py#L257-L279

See output:
https://gist.github.com/EricCousineau-TRI/c8c5ebf28c6c36acdd8c70727848674d

I can workaround by using the following hack:
https://github.com/RobotLocomotion/drake/blob/9a475540c5d844c2b5b2d5b37eca785f0f758882/examples/instability/test/plant_hidden_state_test.py#L167-L170

\cc @xuchenhan-tri @DamrongGuoy w.r.t. other issues (#18236, #18241)
\cc @jwnimmer-tri @sherm1 w.r.t. #18551, #12649

FYI @dmcconachie-tri When we did our rubber duck session last week, I was afraid of stuff like this haha

### Version

d2c7504c49f7fee1cecb1fe82d21c11420b3d9f1

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

sherm1, 2023-01-28:
@amcastro-tri do you know of any hidden state that could account for this?

amcastro-tri, 2023-02-03:
We discussed this with Eric this past week. 
We do not know what could cause this issue. There are a few easy thing to try:
  1. Disable caching.
  2. Switch to TAMSI
  3. Replace MBP for an equivalent hand written LeafSystem. In this case m*vdot = u, with m = 1.0, 1D.
  
The goal is to find the simplest Diagram that still reproduces the failure. Once there, we'd need to re-evaluate.

xuchenhan-tri, 2023-04-11:
I'm going to take a look at this as part of the ongoing MbP tech debt effort.

xuchenhan-tri, 2023-04-12:
The problem is quite intriguing. Thanks for providing the code snippet! I was able to reproduce. The problem goes away when caching is disabled. The fact that the participation of contact visualizer makes a difference seems to indicate that something is wrong in the computation of ContactResults in CompliantContactManager (potentially a hidden state as Eric suggests).

amcastro-tri, 2023-04-12:
I wonder if one of the cache tickets could have the wrong dependencies. 
Something I can think of is declaring a dependency on the kinematics ticket, which is only valid for continuous mode. For discrete mode it should be the discrete state.

xuchenhan-tri, 2023-04-12:
> I wonder if one of the cache tickets could have the wrong dependencies.

You are on the right track. I think the issue is the contact solver cache entry is incorrectly not depending on the actuation input port. I'll PR a fix soon.

xuchenhan-tri, 2023-04-12:
I realize that this is the same issue as #12786. We intentionally not depend on actuation input port and uses caching to emulate a zero order hold, and the landmine blew up in our face again.

xuchenhan-tri, 2023-04-13:
@sherm1 and I had a discussion about this and #19193 is not the desired fix. We don't want the contact solver computation to directly depend on the input ports because for a discrete plant, the input port should also behave discretely.

I believe the cause of @EricCousineau-TRI's issue is something along the line of 
1. The contact visualizer pulls on the generalized contact force output port which computes a value for the contact solver results cache entry.
2. Then the value in MbP's actuation input port changes.
3.  As the contact solver results cache entry only depend on the state and parameters, the cache entry doesn't refresh and the old actuation value is then used to advance the time step.
4. When the `clean_plant_state` flag is turned on, the states are refreshed and therefore the contact solver results are recomputed, using the new actuation values.

The exercise also reveals a defect in an existing unit test, see #19195.

I believe the correct fix is to force an update to the [hack cache entry](https://github.com/RobotLocomotion/drake/blob/530b455bd7a4e00a92875ace1d9477390d996035/multibody/plant/compliant_contact_manager.cc#L132) whenever the discrete update handler is triggered. WDYT? @sherm1 @amcastro-tri

sherm1, 2023-04-13:
IMO it is best to think of this in terms of sampling of discrete input ports (that's the implementation we're trying to emulate with the current hack). So: in a discrete MbP (non-zero time step) we view a call to the discrete update handler as a sampling event. Only at that time should we look at the values of the input ports, which are presumed to have changed. The hack cache entry intentionally fails to note that it is dependent on input ports, to avoid unwanted invalidations. However, when the discrete update handler "samples" the input ports it should also invalidate the hack cache, which will then get recalculated if someone looks at it.

From that I think the right fix is to explicitly invalidate the hack cache at the beginning of the discrete update handler, which would have the effect that Xuchen mentioned above. However ... if the input ports have _not_ changed since the last hack cache evaluation, it will get unnecessarily recalculated which could cause a performance regression. WDYT @amcastro-tri ?

sherm1, 2023-04-13:
The _right_ right fix would be to sample the input ports into matching state variables. Then the cache entry (no longer a hack) would declare its true dependency on both the kinematic state variables and the sampled inputs (rather than the direct inputs which are continuous). TODO: think carefully about exactly when sampling should occur so that we never mix x[n+1] with u[n] when evaluating the cache entry.

xuchenhan-tri, 2023-04-13:
> However ... if the input ports have not changed since the last hack cache evaluation, it will get unnecessarily recalculated which could cause a performance regression.

I wouldn't worry too much about this. The majority of the computation cost occurs inside the calculation of contact solver results (whether the underlying solver is sap or tamsi). Copying the input port values is but a small additional cost.

> The _right_ right fix would be to sample the input ports into matching state variables

As we discussion in f2f, the additional state appears to interfere with other analysis where the user doesn't expect this additional state. I would propose an immediate fix as

1. Using a cache entry to _mimic_ matching discrete input port states as we currently do and documented in #12786.
2. Document that cache entry explicitly as a hack to mimic discrete input ports. Currently that hack is happening in the `non_contact_forces_acceleration` cache entry and is hard to detect.
3. Mimic a sampling of the discrete input at the start of the discrete update handler by forcing a recalculation of the cache entry. We currently are not doing that.

xuchenhan-tri, 2023-04-13:
Per f2f with @amcastro-tri , we will give this proposed solution a try. The cache entry that mimics the discrete input port will have a Calc function that throws when caching is enabled so that when people fall in the same confusing trap as in #12643 when they turn off caching, they at least know what is going wrong.

I'll put up a PR implementing this idea for discussion next week.

PR #19225 (2023-05-02): [multibody] Refresh cache entry hack upon update
Closes #18682.

Systematically adopt the strategy of mimicking discrete input force ports for a discrete MultibodyPlant.
    
1. Ensure that all the discrete sampling happen at a single place.
2. Ensure the cache entry is manually updated at the beginning of a time step.
3. Move all CalcNonContactForce() logic for all MbP discrete updates (that depends on the discretely sampled force input ports) to a single place in DiscreteUpdateManager.
4. Consequently, the ThrowIfNonContactForceInProgress guard against algebraic loops can be centralized in D [... cut]
closed 2023-05-02 by SeanCurtis-TRI