RobotLocomotion/drake #18688: Deprecation removals 2023-02

jwnimmer-tri, 2023-01-28:
The following code is scheduled for deprecation removals on 2023-02-01.

I'm looking for a volunteer to prepare the pull request.  It's probably easiest to do them all at once since they are each pretty small on their own.

From #18179 via @hongkai-dai:
- `drake/common/symbolic/trigonometric_polynomial.h`
- `drake/bindings/pydrake/symbolic_py.cc`
- `drake/bindings/pydrake/test/symbolic_test.py`

From #18094 via @sherm1:
- `drake/systems/framework/system.h`

From #18122 via @amcastro:
- `drake/bindings/pydrake/multibody/tree_py.cc`

rpoyner-tri, 2023-01-31:
I'll take a look and continue the dep-removal theme of my platform review day.

rpoyner-tri, 2023-01-31:
See #18716.

PR #18716 (2023-02-01): Remove deprecated code 2023-02
Closes #18688.

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18716)
closed 2023-02-01 by rpoyner-tri