RobotLocomotion/drake #18704: FCL bug in ComputeSignedDistancePairwisePoints()

SeanCurtis-TRI, 2023-01-31:
### What happened?

A user is doing distance calculations which eventually devolve to a failure in the EPA/GJK algorithm of things not being as expected. The original discussion in [stackoverflow](https://stackoverflow.com/questions/75292185/fcl-error-with-computesigneddistancepairwisepoints-in-pydrake) and the relevant details have been recreated here.

Here's the error message:

```
*** RuntimeError: Error with configuration
  Original error message: external/fcl/include/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1719): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1: Sphere(0.01200000000000000025)
  X_FS1
 -0.40370167831914760548 -0.019479333109567995019   -0.9146832842595900015   0.74747104350671045303
  -0.4289595844695910376   0.88710005183204809764   0.17043231187557600359  0.025509175267072643767
  0.80809568110093199511   0.46116597188125679763  -0.36647962639802561524  0.045779118115638992026
                       0                        0                        0                        1
  Shape 2: Convex(v count: 16, f count: 28)
  X_FS2
  0.98447767053732548881   0.11941117744570119152  -0.12862615175162719905   0.70727978567772809004
 -0.11946133329759119068   0.99281162514245557826 0.0073530149434261633107  0.019925905554638502493
  0.12857957092852342895 0.0081269725622573462526   0.99166589447091968434  0.044593263795921309067
                       0                        0                        0                        1
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```

TODO: Update this problem statement when we have an exact definition of the convex shape. At that point, we should be able to reproduce and debug.

alberthli, 2023-02-01:
Thanks for opening this issue. I don't have the collision geometries associated with the original error message, but I regenerated a different one on a different set of initial conditions.
```
*** RuntimeError: Error with configuration
  Original error message: external/fcl/include/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1719): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1: Sphere(0.01200000000000000025)
  X_FS1
 -0.44891714095597207157   0.86632127192498886714   0.21899966750245522529   0.69134078467990878192
  0.44092752233991133748 0.0015914144089439888948   0.89754130124655895351 -0.047286579335901468557
  0.77721060247471085436    0.4994846556306300478  -0.38269930518678518805   0.11697927955152286061
                       0                        0                        0                        1
  Shape 2: Convex(v count: 16, f count: 28)
  X_FS2
   0.96713841202357886395    0.16339883066460536565   -0.19479248991157169235    0.69861093786591421662
  -0.17593352094622538573    0.98319336270688029167  -0.048766871302155401224 -0.0076967924383244887276
   0.18355023344026075161   0.081434843074549706499    0.97963139911777497026    0.11141130288498900247
                        0                         0                         0                         1
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
```
The object in question has 16 collision geometries and I'm not sure which one is causing the problem. The corresponding obj files are located here:
[collision_meshes.zip](https://github.com/RobotLocomotion/drake/files/10560984/collision_meshes.zip)

I'm using the `jammy-20230112` release in a Docker container. My host machine is running Ubuntu 20.04.5. Let me know what other information would be helpful for debugging.

yongxf-anyware, 2023-02-28:
It seems that this FCL issue persists even for simple box shapes. In my case, The boxes are stacked as a wall, and I try to compute the signed distance between two boxes. The error ```The origin is outside of the polytope. This should already have been identified as separating``` shows up once in a while (~10x per day)

Both the libccd and FCL are installed from the lastest GitHub master branches. 

@SeanCurtis-TRI Any thoughts on how to fix this? 
```
terminate called after throwing an instance of 'std::logic_error'
  what():  Error with configuration
  Original error message: /home/fan/code/anyware-libraries/build/anyware_fcl-prefix/src/anyware_fcl/include/fcl/narrowphase/detail/convexity_based_algorithm/gjk_libccd-inl.h:(1719): validateNearestFeatureOfPolytopeBeingEdge(): The origin is outside of the polytope. This should already have been identified as separating.
  Shape 1: Box0.2999999999999999889 0.2999999999999999889 0.2999999999999999889
  X_FS1
   0.99983216350830295127   0.017302258779861117571  0.0060230104948389619895    0.85626818476862631702
 -0.017255387167394182352    0.99982108658241231058 -0.0077489637286347464937   -0.77764714060125894601
-0.0061560074731557591632   0.007643733791746519686     0.9999518372930324972   0.070907056388150049853
                        0                         0                         0                         1
  Shape 2: Box0.2999999999999999889 0.2999999999999999889 0.2999999999999999889
  X_FS2
   0.99985651556315335853   0.016542340539402608512  0.0036468144164748281071    0.86027781426343519211
 -0.016512132944187574157    0.99983034011987759193 -0.0081633596884769909358   -0.76973597919045266824
 -0.003781236774290517872  0.0081019716889417000422    0.99996002935277783497   -0.22878728666065184516
                        0                         0                         0                         1
  Solver: GjkSolver_libccd
    collision_tolerance:      2.0097183471152322134e-14
    max collision iterations: 500
    distance tolerance:       9.9999999999999995475e-07
    max distance iterations:  1000
Aborted (core dumped)
```

SeanCurtis-TRI, 2023-02-28:
@yongxf-anyware the fact that it's between two boxes makes this easier to track down. In this case, the error dump *should* provide sufficient information to reproduce and we can see what subtle bug this reveals.  The EPA stage is fraught with numerical subtleties and we've taken to trying to knock them out, one at a time.

I'll try playing with this dump and get back to you.

SeanCurtis-TRI, 2023-02-28:
@yongxf-anyware Frustratingly enough, I'm unable to reproduce the bug -- see [this code in a branch of mine](https://github.com/SeanCurtis-TRI/fcl/blob/PR_gjk_epa_regression7/test/test_fcl_signed_distance.cpp#L479-L498).

So, it makes me wonder if there's something extra going on.

1. What version of drake do you have?
2. Do you have FCL on your system that *isn't* the one Drake manages? What version are you running?

jwnimmer-tri, 2023-02-28:
The file paths in the error messages tell a story.

The messages from `@alberthli` show a Drake build of FCL.  Those might be worth pursuing.

The message from `@yongxf-anyware` shows a non-Drake build of FCL (being called from non-Drake code).  So this one isn't a Drake issue, and is out of scope here.

SeanCurtis-TRI, 2023-02-28:
An excellent point! It's just a shame that the convex isn't as nice to deal with as the Boxes. I should probably augment the error messages so that failure with a convex shape outputs sufficient information to reproduce.

yongxf-anyware, 2023-02-28:
Thank you both @SeanCurtis-TRI and @jwnimmer-tri. I am not using Drake indeed. The reason I replied in this thread is that @SeanCurtis-TRI contributes a lot to this FCL epa issue in general. I thought my case may be a good data point. 

I am using [FCL 0.7.0](https://github.com/flexible-collision-library/fcl/tree/0.7.0) installed from the source. It has the same 
 `gjk_libccd-inl.h` as your gjk_libccd-inl.h. 

The associated libccd is also the [latest version](https://github.com/danfis/libccd/tree/7931e764a19ef6b21b443376c699bbc9c6d4fba8) and installed from source. Yet, I am using -DENABLE_DOUBLE_PRECISION=OFF since -DENABLE_DOUBLE_PRECISION=ON makes my own collision detection `stack smashing`. 

I will ask for solution under FCL github later. Thanks for pointing out!

SeanCurtis-TRI, 2023-03-02:
@alberthli 

I've now spent some cycles on this, but haven't had any success:

1. I've tried reproducing the errors reported above with the objs linked above. My efforts have failed. You can see my efforts in [this branch](https://github.com/SeanCurtis-TRI/drake/tree/PR_signed_distance_failure_18704). You can run the test program as:

    `bazel run //examples/scene_graph:run_signed_distance`

    Running all objs against the spheres produces measurements with no exceptions, so, clearly, something is missing.

2. I've submitted a modification to fcl (flexible-collision-library/fcl#594). The goal is to have the error message output more information (even for `Convex`) so that we can reproduce errors without having to fish for files. That will have to merge in fcl and Drake will have to update its reference before you can see the benefit. I'll let you know when it is available.

3. I'm trying to figure out what might be missing. Assuming that one of the OBJ files *must* be the offender, I'm force to turn to platform. What machine are you running on? Mac (and what kind) or Ubunutu (and what OS)?

alberthli, 2023-03-02:
Thanks for your efforts, @SeanCurtis-TRI.

I'm running Ubuntu 20.04 on a Lenovo Thinkpad X1 Carbon Gen 8 and running Drake using the `jammy-20230112` release in a Docker container (i.e., the container is running 22.04). Let me know if there's more fine-grained information I can provide. I can devote some time next week to trying to cook up a more reproducible minimal example, since it seems the object files themselves aren't enough.

SeanCurtis-TRI, 2023-03-02:
I'd stall just a bit until the FCL PR lands. Life will be simpler after that. (Unless you happen across an easily reproducible case, then, of course, I'd happily accept that.) But, for now, probably not worth taking any extraordinary effort.

As for platform, that all sounds reasonable (unhelpfully so). I'll have to think of something else.

alberthli, 2023-03-02:
Got it. I'll keep an eye out for the FCL PR.

SeanCurtis-TRI, 2023-03-15:
@alberthli #19006 finally percolated all the way through to Drake (FCL had grown moribund in the last two years). 

I'm going to close this issue for now. When you have the chance to pull the most current Drake (sadly it didn't get into v1.14.0), if you have a repeat of the problem, you should get a new happy message that has more data. Feel free to open a new issue with the new data.

SeanCurtis-TRI, 2023-03-15:
FYI The critical change in FCL was https://github.com/flexible-collision-library/fcl/pull/594.

closed 2023-03-15 by SeanCurtis-TRI