RobotLocomotion/drake #18705: Wheel build local testing on Jammy

jwnimmer-tri, 2023-01-31:
In CI, we build all of our manylinux wheels using Docker running on a Focal host system.

Ideally, we'd like Drake Developers be able to locally build wheels using Docker, but running on a Jammy host system instead.  (Most developers use Jammy at this point.)

The only thing stopping that at the moment is that in the drake source tree for Jammy, the file `drake/gen/environment.bazelrc` refers to Jammy ...

```
import %workspace%/tools/ubuntu-jammy.bazelrc
```

... because that's how the source tree was configured for local development (via `setup/ubuntu/source_distribution/install_prereqs_user_environment.sh`).

In the wheel build, we somehow need to switch the file to say **focal** instead:

```
import %workspace%/tools/ubuntu-focal.bazelrc
```

Possibly the `_create_source_tar` linux step could replace that one file during the `tarfile` operation?  Notably, it's the _only_ file in the `gen/` directory in the tarball, so hopefully should be easy to special-case.

mwoehlke-kitware, 2023-01-31:
...maybe the wheel build should just assume it needs to recreate that?

jwnimmer-tri, 2023-01-31:
We shouldn't change the user's file in their own source tree; that would break their other non-wheel builds.  The fix should happen either in the tarball creation, or else via the within-container scripts, or some mix of the two.

mwoehlke-kitware, 2023-01-31:
That's what I (thought I) said: the wheel script *run in the docker container* should assume it can't trust the file that came from the tarball. Sorry, it wasn't clear I was talking about that happening *in the docker container*. (Might not be a bad idea to exclude that file from the tarball, also. Should make for easier testing, at least.)

...So it sounds like we're on the same page?

jwnimmer-tri, 2023-01-31:
Sure, seems fine.  My thought was that fixing it in `linux.py` would be easiest since it knows which distro the container is about to use.  Fixing it within the container scripting should also work.  Whatever is easiest.

(FYI note that this ticket is way low on the board priorities, though.)

jwnimmer-tri, 2023-10-05:
Fixed in #18999, and then fixed even harder in #20311.

closed 2023-10-05 by jwnimmer-tri