RobotLocomotion/drake #18711: [solvers] Possible OSQP non-determism?

EricCousineau-TRI, 2023-01-31:
### What happened?

Building and solving the same `MathematicalProgram` can produce different results.

See: https://github.com/EricCousineau-TRI/repro/tree/5f0716147d62c1769d8b3b77a61a51ff1ef4418e/drake_stuff/issues/qp_nondetermism

Slack convo:
https://drakedevelopers.slack.com/archives/C3L92BM2Q/p1675184284394289

### Version

1.12.0

### What operating system are you using?

Ubuntu 22.04

jwnimmer-tri, 2023-01-31:
One tip if we decide to do `CommonSolverOptions::kDeterministic` then we should search each solvers' options for use of "time" or "clock" but also "rand" or "guess".  I think timing and randomness are the most common two sources of non-determinism.

jwnimmer-tri, 2023-01-31:
Yes, I think no matter what we choose for the built-in defaults, any options that the user provides in their `SolverOptions` should supersede the defaults.

abhishek-pandala, 2023-02-01:
Hi everyone, This is a possible suggestion and feature request. I noticed that drake supports only OSQP as its primary QP solver. There are many other [qp solvers](https://github.com/stephane-caron/qpsolvers#solvers) available. Is there a reason not to choose them?

I myself am a developer of qpswift and would like to see if it is possible to integrate qpswift with drake and get some feedback. Here is a link for [qpswift repo](https://github.com/qpSWIFT/qpSWIFT) and some of the legged robots where it is successfully [deployed](https://github.com/qpSWIFT/qpSWIFT/wiki/6.Applications-on-Robots)

qpswift uses a Primal-Dual Interior point method and generates accurate solutions as it uses a second order method.

jwnimmer-tri, 2023-02-01:
@abhishek-pandala thanks for your interest!

We should probably open a new feature request issue number to discuss (since this issue numer is focused on OSQP) -- you can do that [here](https://github.com/RobotLocomotion/drake/issues/new/choose) -- but I can give you a quick answer for the moment:

We are always interested in hearing about new solvers, and are curious to try them.  Thanks for the list.

Taking a quick look at qpSWIFT, it seems to be offered under the GPL license, which makes it incompatible with Drake.  We have a wide audience of users of Drake and not all of them are capable of fulfilling responsibilities that come along with the GPL.

You might consider re-licensing qpSWIFT under [MPL-2.0](https://spdx.org/licenses/MPL-2.0.html), which is used by some other projects, e.g., [Eigen](https://eigen.tuxfamily.org/index.php?title=Main_Page).  Like the GPL, it requires that any improvements to qpSWIFT must be redistributed, but does not force disclosure of all of the source code for rest of the unrelated software compiled into the same program.

If that's of interest to you, free free to open a new Drake feature request on the topic and we can take it from there.

jwnimmer-tri, 2024-03-17:
I re-titled the issue to choose Option 1 as the call to action (default `adaptive_rho` to zero, unless the user provides a more specific value).  Drake should not be wallclock-sensitive by default.

jwnimmer-tri, 2024-05-15:
Reading more about `adaptive_rho` now in the upstream source code and issues database (e.g., https://github.com/osqp/osqp/issues/235#issuecomment-585420663), my new vote is that we should leave `adaptive_rho=true` by default.

Instead, we should default to `adaptive_rho_interval=ADAPTIVE_RHO_FIXED` (i.e., `adaptive_rho_interval=100`) as our Drake-specific option, instead of the upstream default of `adaptive_rho_interval=0`.  The default of zero means "latch-initialize the choice of interval based on observed wallclock runtimes".

RussTedrake, 2024-05-15:
I agree. Do we have any sense for how this impacts solver performance?

bstellato, 2024-05-15:
I second @jwnimmer-tri's suggestion. (cc'ing @ianmcinern

The nondeterminism comes from the wallclock time estimation in the setup phase. The reason we included that is the following. If the time to run the ADMM iterations starts to be close to the one of your initial factorization, you may as well update `rho` and factor the KKT matrix again. We also explain this more in detail in the [OSQP paper](https://web.stanford.edu/~boyd/papers/pdf/osqp.pdf):

> Since the numerical factorization can be costly, we perform the adaptation
only when it is really necessary. In particular, we allow an update if the accumulated
iterations time is greater than a certain percentage of the factorization time (nominally
40%) and if the new parameter is sufficiently different than the current one, i.e., 5 times
larger or smaller.

My suggestion is to have an adaptive `rho` schedule (`adaptive_rho=1`), and to do it at fixed time intervals so that determinism is ensured. This could be done, for example, with `adaptive_rho_interval=50` (this may be a tuning parameter). There is also a discussion here https://github.com/osqp/osqp/issues/583.

jwnimmer-tri, 2024-05-15:
Thanks!

FYI my specific proposal is PR #21434 now, and includes some user-directed advice in comments.

Thanks for the link to https://github.com/osqp/osqp/issues/583, that's right on point. I'll add that link to the code comments in the next push the PR.

PR #21434 (2024-05-15): [solvers] Put OsqpSolver in deterministic mode by default
When adaptive_rho is enabled (the default) and adaptive_rho_interval is zero (the default), OSQP will guesstimate a suitable interval using wall clock timing measurements. This leads to non-reproducible results based on changes in CPU timing. In Drake we treasure reproducibility, so we should set our own default using a fixed schedule. We'll choose adaptive_rho_interval=ADAPTIVE_RHO_FIXED, which is OSQP's default value on embedded platforms that lack access to timing measurements.

Users who wish to keep using adaptive_rho_interval=0 ca [... cut]
closed 2024-05-15 by hongkai-dai