RobotLocomotion/drake #18730: Drake should support planar joints in URDF under the standard `joint` tag

rpoyner-tri, 2023-02-02:
### What happened?

An URDF file with a planar joint failed to parse, with the error:

```
Joint joint0 of type planar is a custom joint type, and should be a <drake:joint>
```
but, in fact, `planar` is a standard joint type in URDF.

Slack discussion here: https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1675340885355229

### Version

v1.11+

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

SeanCurtis-TRI, 2023-02-06:
@rpoyner-tri I've assigned this to you as resident parser king. Feel free to pass it along as appropriate.

SeanCurtis-TRI, 2023-02-06:
Did #18735 close this?

xuchenhan-tri, 2023-05-01:
#18735 closes this but somehow the github keyword didn't kick in.

closed 2023-05-01 by xuchenhan-tri

jwnimmer-tri, 2023-05-02:
(The magic has to be in the PR description (top post) for it to have the auto-close effect.)