RobotLocomotion/drake #18732: Add getters to interrogate existing constraints

rpoyner-tri, 2023-02-02:
Currently MultibodyPlant does not provide any access methods to examine existing constraint specs objects that have been added.  The immediate pain point arises in writing unit tests, but surely users will also benefit.

Proposal: add public methods to MultibodyPlant that enable interrogation of constraint specs objects. In addition, expose or re-implement the parameter type information that is currently in the `internal` namespace to allow access to specific parameters of the various constraint types.

rpoyner-tri, 2023-02-02:
\cc @amcastro-tri

amcastro-tri, 2023-02-03:
Thus far specs are `internal::`. If a user only wants to "read back" specs provided earlier, making these spec objects public API could get the job done.