RobotLocomotion/drake #18740: [py value] Using copy.deepcopy for PyValueObject makes writing a simple Python logger painful

EricCousineau-TRI, 2023-02-04:
**Is your feature request related to a problem? Please describe.**
The PR #18655 was meant to resolve #18653 by using `copy.deepcopy` throughout to maintain value semantics.

This objective was achieved; however, it now makes writing / prototyping a simple logger as shown below very slow and/or painful. For prototyping controllers + haptic coupling, a diagram that used to run real time (~500 Hz) now runs much slower due to the effective calls to  `copy.deepcopy(log)`, where `len(log) ~= int(context.get_time() / dt)`.

**Describe the solution you'd like**
Either a clear workaround, or an option to avoid deep copying when running against a "forward-moving-only" context (e.g. as a result of calling updates against the same context in succession, such as `Simulator.AdvanceTo`).

**Describe alternatives you've considered**
For the workaround, I'm trying to figure out stupid (maybe robust?) way to make something like `Value[MyToken]`, where the token is unique for "forward movement".

For now, registration / update will look like storing the key `(id(token), context.get_time())`, and the token becomes invalid (somehow) when the stored time is not within the bounds of `[t_prev + 0.5 * dt, t_prev + dt * 1.5]`.

**Additional context**
The following logger can show the issue, especially if each data point is more than a simple vector (e.g. `RigidTransform`, images, etc.):
```py
class ScalarLog(LeafSystem):
    def __init__(self, dt):
        super().__init__()

        log_init = list()
        log_index = self.DeclareAbstractState(Value[object](log_init))
        DeclareSomeSortOfInputPort(...)

        def eval_log(context):
            context = self.GetMyContextFromRoot(context)
            log = context.get_abstract_state(log_index).get_value()
            return log

        self.eval_log = eval_log

        def eval_mutable_log(context, raw_state):
            abstract_state = raw_state.get_mutable_abstract_state(log_index)
            log = abstract_state.get_mutable_value()
            return log

        def on_discrete_update(context, raw_state):
            t = context.get_time()
            log = eval_mutable_log(context, raw_state)
            u = self.get_input_port().Eval(context)
            log.append((t, u))

        self.DeclarePeriodicUnrestrictedUpdateEvent(
            period_sec=dt,
            offset_sec=0.0,
            update=on_discrete_update,
        )
```

EricCousineau-TRI, 2023-02-04:
\cc @jwnimmer-tri @sherm1 if y'all have any ideas.

EricCousineau-TRI, 2023-02-04:
Found workaround; "blech", but also "meh, it seems to work"

```py
_CONTEXTS = set()

def _stablize_context_id(context):
    # Blech memory leak :( Oh well.
    # TODO(eric.cousineau): Add uuid mechanism to context? e.g. bind something
    # stupid like `context.get_uuid()` return runs `int(this)`?
    _CONTEXTS.add(context)

@dc.dataclass
class TokenState:
    uuid: int
    context_id: int
    context_time: float

    @staticmethod
    def make_new(context):
        _stablize_context_id(context)
        return TokenState(
            uuid=id(object()),
            context_id=id(context),
            context_time=context.get_time(),
        )

    @staticmethod
    def is_valid(token, context):
        _stablize_context_id(context)
        return (
            token is not None and
            id(context) == token.context_id and
            context.get_time() >= token.context_time
        )

    def update(self, context):
        _stablize_context_id(context)
        assert TokenState.is_valid(self, context)
        self.context_time = context.get_time()

class TokenizedSystemState:
    # WARNING: Use `log.append()` directly in Python will slow things way
    # down. It is better to simply store stuff outside.
    # See https://github.com/RobotLocomotion/drake/issues/18740

    def __init__(self, system, default):
        self._state_index = system.DeclareAbstractState(Value[object](None))
        self._make_default = lambda: copy.deepcopy(default)
        # TODO(eric.cousineau): How to clean up? Meh for now.
        self._memory = {}

    def eval(self, context):
        token = context.get_abstract_state(self._state_index).get_value()
        if TokenState.is_valid(token, context):
            return self._memory[token.uuid]
        else:
            # Don't make a new one since we can't update the token id.
            return self._make_default()

    def update_and_eval_mutable(self, context, raw_state):
        # N.B. We don't really need mutation, and perhaps we don't even
        # need to use mutable accessors because calling setup will dirty
        # the caching.
        abstract_state = raw_state.get_mutable_abstract_state(self._state_index)
        token = abstract_state.get_value()
        if TokenState.is_valid(token, context):
            token.update(context)
            value = self._memory[token.uuid]
        else:
            token = TokenState.make_new(context)
            value = self._make_default()
            self._memory[token.uuid] = value
        abstract_state.set_value(token)
        return value

# Change above logger as follows
class SimpleLogger(LeafSystem):
    def __init__(self, dt):
        super().__init__()
        ...

        log_state = TokenizedSystemState(self, [])

        def eval_log(context):
            return log_state.eval(context)

        def eval_mutable_log(context, raw_state):
            return log_state.update_and_eval_mutable(context, raw_state)

        ...
```

EricCousineau-TRI, 2023-02-04:
Ah, I see that `VectorLogSink` uses caching:
https://github.com/RobotLocomotion/drake/blob/fa26b17be5d716b3302741e9e659f6910b242ed8/systems/primitives/vector_log_sink.h#L29-L33

Well, I could do that!
However, this approach allows for use with `SetStateAndParameters`, which I don't believe using cache will allow.

jwnimmer-tri, 2023-02-04:
Having a logger that accumulates its data into State is where you went wrong.  A log doesn't have dynamics, nor affect any outputs -- it is not State.

We _do_ actually need deep copies of State.  In cases where its frequently being copied without ever changing it, then choosing an abstract value type that has CoW semantics can help in other cases, but here it looks like it's being modified a lot anyway, so that wouldn't even help.

If you your need for logging is simple enough that the log can be System-specific instead of Context-specific, then you can log into `self._private_data`.  That works in simple cases where you have only one Context per system.

If you do need context-specific logging, then either use a CacheEntry like the C++ logger does, or else log into `self._private_data` on the System but store it as a dictionary keyed on Context so that each Context has separate log data.

jwnimmer-tri, 2023-02-04:
The other option is to use a value type other than `list[]`.  Does Python have an immutable, rope-like sequence class?  If there were a sequence class where deep copying was a no-op because it was immutable, but it offered `append()` to return a new object O(1) that reference the old object, that would work here.

jwnimmer-tri, 2023-02-04:
Here's an example of a cheap-copy logger class.

The update event would to something like:

```py
old_log = context.get_abstract_state(log_index).get_value()
new_log = old_log.appended((t, u))
next_state.get_mutable_abstract_state(log_index).set_value(new_log)
```

```py
import copy

class Log:
    def __init__(self):
        self._values = ()
        self._ancestor = None

    def appended(self, value):
        result = Log()
        result._values = (value,)
        result._ancestor = self
        return result

    def as_tuple(self):
        result = []
        iterator = self
        while iterator is not None:
            result[0:0] = iterator._values
            iterator = iterator._ancestor
        return tuple(result)

    def __copy__(self):
        return self

    def __deepcopy__(self, _):
        return self

log = Log()
for i in range(100000):
    # log = copy.deepcopy(log)
    log = log.appended(i)
print(len(log.as_tuple()))
```

jwnimmer-tri, 2024-03-16:
> > Describe the solution you'd like
>
> Either a clear workaround, ...

My prior comment showed how to use an efficient, append-only `Value[Log]`.

closed 2024-03-16 by jwnimmer-tri