RobotLocomotion/drake #18741: Make it easier to add models with a fixed base (welded to the world)

RussTedrake, 2023-02-04:
Other simulators [like bullet](https://github.com/bulletphysics/bullet3/blob/master/docs/pybullet_quickstart_guide/PyBulletQuickstartGuide.md.html#loadurdf) have an option to `useFixedBase` when they load a model.  

The equivalent incantation in Drake is more cumbersome, especially if you want to write it in a way that doesn't require knowing the string name of the base body. This is especially true since `GetFloatingBaseBodies` isn't avaible pre-finalize (and doesn't take a model index).  I suppose this could work for most models, but it assumes that the first body is the base body:
```
  models = parser.AddModels(robot_drake_paths[args.robot])
  for m in models:
      base_body_index = plant.GetBodyIndices(m)[0]
      plant.WeldFrames(plant.world_frame(), plant.get_body(base_body_index).body_frame())
```

RussTedrake, 2023-05-15:
The victory condition I was imagining was that by the time parsing is complete, all bodies in the (arbitrary) urdf/sdf/xml/dmd.yaml file that would have had a floating base are instead welded to the world origin.  We currently do support model authors adding the weld joint manually in urdf/sdf/xml in addition to dmd.yaml; the request is that they do not have to annotate the weld.

jwnimmer-tri, 2023-05-15:
I'm not completely clear yet on why it's useful to weld things at xyzrpy=0,0,0,0,0,0.  Grepping around in Anzu for existing welds to the world with no offset, it seems like it's used most frequently when loading a robot arm into relatively simple scenarios -- we weld the arm's base link to the world (no offset), and then try to grasp some object on the ground plane or something.  Does that match up with the kinds of cases you are finding that would want this kind of unparameterized welding?

I ask because if there are two things being welded, surely the user would want to have an offset for the welds, using specifically named weld joint offsets and/or offset frames. Otherwise (without offsets), everything would smoosh together at the origin. So this sugar would only be used once per scenario, not e.g. on every manipuland.

Even when welding just one robot to the world with no offset in the weld, in Anzu we frequently weld to a specific frame within the model, other than the first link.  (The URDF origin of the first link is not always a natural reference point when physically bolting together the hardware.)  This shortcut would not support those use cases, because it wouldn't know about the "weld here please" frame name.

In any case, perhaps for the simple cases you're looking at a little dmd sugar would be sufficiently simple?

Instead of saying this:

```yaml
directives:
- add_model:
    name: panda
    file: package://drake/manipulation/models/franka_description/urdf/panda_arm.urdf
- add_weld:
    parent: world
    child: panda::panda_link0
```

The user could say this:

```yaml
directives:
- add_model:
    name: panda
    file: package://drake/manipulation/models/franka_description/urdf/panda_arm.urdf
    static: true
```

The `static: true` would de-sugar to an AddWeld between the world and the canonical frame in each model in the file.  I still think writing out the AddWeld is good for the soul (so it's ultra-clear what is going on, and so that we have a natural place to change the default static pose and/or use offset frames), but we could imagine adding this as sugar if people really do want to weld things at 0,0,0,0,0,0 frequently.

sherm1, 2023-05-15:
BTW the new MbP model building code I'm working on supports "static" links which get welds rather than floating joints, and for floating joints whether to use quaternions or rpy. Would be easy to have an option to use welds all around instead.

RussTedrake, 2023-05-16:
@jwnimmer-tri -- yes, I do think that the most common use case is simple setups where a single robot is welded to the origin. I agree that it would be more strange if there are multiple floating bases in the parsed file; my recommendation for welding all of the floating bases was more because I feel that it is a more reasonable behavior than to arbitrarily pick the first one and seems less friendly to throw if there are multiple instances.

The bullet interface (linked above) allows users to specify the base position and orientation, too.  So I agree it is almost the weld joint that we have in model directives or via plant.WeldFrames. The more important part of the sugar, I think, is not the lack of specifying the pose, but the lack of needing to name the body properly.

Perhaps the need could be met if we added a method like `MultibodyPlant<T>::FixBase(model_instance, X_WB=RigidTransform())`?

jwnimmer-tri, 2023-05-16:
> The more important part of the sugar, I think, is not the lack of specifying the pose, but the lack of needing to name the body properly.

Okay, we can focus on that part.

FYI If the robot comes from SDFormat, you can already do this to weld the canonical link (usually, the first link) to the world:

```yaml
directives:
- add_model:
    name: panda
    file: package://.../panda_arm.sdf
- add_weld:
    parent: world
    child: panda::__model__
```

Plausibly we could update our other parsers (beyond SDFormat) to also add a `__model__` frame to their models, so that the same syntax would work for all models.

I'll also think about the MbP function you proposed and related solutions.

If the graph building (finalize workflow) can configure the welds, that might be a good solution too.

sherm1, 2024-11-11:
MbP in Drake master now has a C++ API to control the kind of joint used to attach a base to World: quaternion floating, rpy floating, or weld. See [SetBaseBodyJointType()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a7c9672b405425259b192131304f14f9c)

rpoyner-tri, 2026-11-10:
Having heard no updates for 2 years, closing.

closed 2026-11-10 by rpoyner-tri