RobotLocomotion/drake #18743: Configurable Lighting in Meshcat

yu-fz, 2023-02-05:
I was looking at the drake meshcat API to see if I could customize the default meshcat scene lighting. The Meshcat class [documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_meshcat.html#af3dc13928478563343ff6ef72c16fc98) suggests that adding lights is possible: `You can modify these elements, create new lights/cameras, and even delete these elements (one at a time)`. 
However, a look through the meshcat methods and the meshcat js [source code ](https://github.com/rdeits/meshcat/blob/master/src/index.js) suggests that this may not be implemented. I can change certain parameters for the default scene lights in meshcat such as `intensity`,`position`, and `castShadow` via a `meshcat.SetProperty(...)` call, but it seems that I am unable to change the light color or add any new lights. 

A related issue is that the meshcat path `/Lights/DirectionalLight` appears deprecated, but this is still provided as an example for the meshcat.SetProperty() documentation, and has been remapped to `FillLight` instead. 

Suggested solution: Provide a SetLight() method for meshcat that takes as its arguments a path, light type, RGB color, intensity, and position. Any additional properties specific to certain light types such as `decay`, `distance`, `castShadow` can be modified after construction via `meshcat.SetProperty()`

SeanCurtis-TRI, 2023-09-12:
I've been poking into this as well, and generally agree.

I'd take this a step further and suggest that we need to add light configuration to a meshcat configuration struct (and a corresponding yaml) so that the lights can be configured easily. That configuration struct should also be consumed by `model_visualizer` and `meldis`.

jwnimmer-tri, 2024-02-13:
> I can change certain parameters for the default scene lights in meshcat such as `intensity`,`position`, and `castShadow` via a `meshcat.SetProperty(...)` call, but it seems that I am unable to change the light color or add any new lights. 

FYI see https://github.com/RobotLocomotion/drake/pull/20920, where the user can set properties at construction-time now.  As you say, that only allows edits to the existing lights, not adding new ones.

For adding new lights, I recommend this approach:
https://github.com/RobotLocomotion/drake/issues/20884#issuecomment-1928768453

If we allow an advanced user to call `Meshcat::SetObject(path : str, object_json : str)` with an arbitrary JSON object, then the user can place whatever kind of light they want into that description.  That's a kind of general solution that unlocks a lot more use cases than just this one.

Of course it would also be a kindness to add "syntactic sugar" to our Meshcat class where the user can specifically add a light and we have nice kwargs to choose its settings, but at the end of the day it just desugars down to a `set_object` call.

jwnimmer-tri, 2024-11-02:
Closing as duplicate of #20941.

After we implement #20941, if we find that things are still too difficult we can circle back here and re-open / file a new ticket.

closed 2024-11-02 by jwnimmer-tri