RobotLocomotion/drake #18745: CI Failure: `internal_render_engine_vtk_test` sometimes fails on Mac Arm

BetsyMcPhail, 2023-02-06:
### What happened?

`//geometry/render_vtk:internal_render_engine_vtk_test` has failed on the following jobs:

- [mac-arm-monterey-clang-bazel-nightly-debug](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-clang-bazel-nightly-debug/76) Jan 24
- [mac-arm-monterey-clang-bazel-nightly-everything-release](https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-clang-bazel-nightly-everything-release/112)  Feb 4
- [mac-arm-monterey-clang-bazel-nightly-debug]( https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-clang-bazel-nightly-debug/90/) Feb 5

### What operating system are you using?

macOS 12 (Monterey)

### Relevant log output

```shell
[2023-02-05 08:09:36.665] [console] [warning] Texture map '/Users/admin/workspace/mac-arm-monterey-clang-bazel-nightly-debug/_bazel_admin/c321c5db3bfe1b41e5a3a7639d47261f/execroot/drake/bazel-out/darwin_arm64-dbg/bin/geometry/render_vtk/internal_render_engine_vtk_test.runfiles/drake/geometry/render/test/meshes/box16u.png' has an unsupported bit depth, casting it to uchar channels.
geometry/render_vtk/test/internal_render_engine_vtk_test.cc:1013: Failure
Expected equality of these values:
  color_uchar_texture
    Which is: 32-byte object <80-02 00-00 E0-01 00-00 00-00 25-38 01-00 00-00 00-C0 37-38 01-00 00-00 00-C0 37-38 01-00 00-00>
  color_uint16_texture
    Which is: 32-byte object <80-02 00-00 E0-01 00-00 00-80 FD-10 01-00 00-00 00-40 10-11 01-00 00-00 00-40 10-11 01-00 00-00>
[  FAILED  ] RenderEngineVtkTest.NonUcharChannelTextures (2288 ms)
```

ggould-tri, 2023-05-16:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/222/consoleFull Again.

jwnimmer-tri, 2023-05-25:
If such flakes are taking too much buildcop attention, I think it would be OK to mark the test as `flaky = True`.

Or, since we already have about 35% of the test cases disabled on Arm already, you could add `RenderEngineVtkTest.NonUcharChannelTextures` to the nerf list as well.

svenevs, 2023-06-05:
Two more instances on June 4:

- https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-unprovisioned-clang-bazel-nightly-release/246/consoleFull
- https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-clang-bazel-nightly-debug/214/consoleFull

BetsyMcPhail, 2023-06-05:
I agree, working on a PR now

PR #19530 (2023-06-05): Disable RenderEngineVtkTest.NonUcharChannelTextures test on Mac Arm
Resolves #18745

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19530)