RobotLocomotion/drake #18756: Provide warning when MultibodyPlant.SetDefaultPositions() is set outside position limits

yu-fz, 2023-02-08:
Currently, MultibodyPlant.SetDefaultPositions() will set joint positions directly, without first checking the position limits of the joint. 
This may cause MbPs to initialize in bad states if the user is not careful about selecting initial positions that respect the corresponding limits prescribed by model files or set elsewhere. 

### Suggestion: 
When `SetDefaultPositions()` sets joint positions, check to see if `q` is between `Joint.position_lower_limits()` and `Joint.position_upper_limits()`. Provide a warning message to the user in the event of any position limit violations, including the joint name and the upper and lower limit values in the warning message.

amcastro-tri, 2023-02-15:
I wonder, is a warning message preferable over an exception?

sherm1, 2023-02-15:
Closing as duplicate of #15765 where there is a useful discussion regarding subtleties of this request.

closed 2023-02-15 by sherm1