RobotLocomotion/drake #18757: numpy warnings on symbolic variables are back?

RussTedrake, 2023-02-09:
### What happened?

```
import numpy as np
from pydrake.all import Variable, SymbolicVectorSystem

q = Variable('q')
print(np.sin(q))
print(np.__version__)
```
results in 
```
sin(q)
1.24.1
[/var/folders/6p/t1z4stds49974ltdtn62_gzw0000gr/T/ipykernel_11472/4207874814.py:5](https://file+.vscode-resource.vscode-cdn.net/var/folders/6p/t1z4stds49974ltdtn62_gzw0000gr/T/ipykernel_11472/4207874814.py:5): RuntimeWarning: invalid value encountered in sin
  print(np.sin(q))
```

### Version

master

### What operating system are you using?

macOS 12 (Monterey)

### What installation option are you using?

compiled from source code using Bazel

rpoyner-tri, 2026-11-11:
I can no longer reproduce this. With the Drake 1.58.0 release wheel and NumPy 2.5.4 (Python 3.12, Linux), `np.sin(Variable('q'))` prints `sin(q)` with no `RuntimeWarning`, even with warnings set to "always". I haven't tried it on macOS.

Closing; please reopen if you still see it on a current NumPy.

rpoyner-tri, 2026-11-11:
Correction to my earlier comment: that test was invalid, and I'm reopening this.

I set `warnings.simplefilter("always")` before `import pydrake`. Importing pydrake installs an `ignore` filter for "invalid value encountered in" (the #18017 hotfix) that goes ahead of mine, so the warning was hidden and I wrongly concluded it was gone.

With the filter set after the import (Drake 1.58.0 release wheel, NumPy 2.5.4, Python 3.12, Linux), `np.sin(Variable('q'))` still emits `RuntimeWarning: invalid value encountered in sin`.

So the underlying cause is not fixed. Drake only hides the warning by default, and it still shows up when a user's own warning filters are applied after the import, which may be what happens in a notebook.