RobotLocomotion/drake #18762: tags = [

svenevs, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

svenevs, 2023-02-09:
@jwnimmer-tri when you are back we may need your help triaging this.

jwnimmer-tri, 2023-02-10:
Is there still any relevant work here?  If CI is passing now (by fixing the test), it seems like we can close this and revisit only if it comes up again.

svenevs, 2023-02-10:
Happy to close, CI is OK for now.  However, it's likely this is going to come back up again in the future.  Linux correctly skips the tests, it seems really problematic that we even had the CI failures in the first place.  So if there is an action item for this ticket, it's "identify why the test was running" and then subsequently remedy that.

closed 2023-02-10 by svenevs

jwnimmer-tri, 2023-02-10:
Yup.  FYI there is https://stackoverflow.com/questions/7208070/googletest-how-to-skip-a-test another way to skip googletests.  Also `args = ["--test_filter=..."]` is another option.