RobotLocomotion/drake #18766: PS5 Controller not detected by meshcat

adeeb10abbas, 2023-02-10:
### What happened?

When using PS5 controller with meshcat on the example [here](https://github.com/Zhourobotics/outdoor_slam_ros2/blob/main/planning/teleop/punito.py), I get `No gamepad found.  Using keyboard/slider teleop.` even though the controller is detected in the following gamepad testers - 
a - https://gamepad-tester.com/
b - https://beej.us/blog/data/javascript-gamepad/ 
Please see the attached video as a demo of this issue. 

https://user-images.githubusercontent.com/38449494/218222602-8edd4c5a-2c29-4404-a0ab-4550e3b14227.mov

### Version

153195c80cc35ca7dfee91dca3c7e76238708434

### What operating system are you using?

Other

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
INFO: Analyzed target //planning/teleop:punito (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //planning/teleop:punito up-to-date:
  bazel-bin/planning/teleop/punito
INFO: Elapsed time: 0.293s, Critical Path: 0.00s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Running command line: bazel-bin/planning/teleop/punito
INFO:drake:Meshcat listening for connections at http://localhost:7001
No gamepad found.  Using keyboard/slider teleop.
```

adeeb10abbas, 2023-02-10:
OS: macOS 13 

/cc: @RussTedrake

jwnimmer-tri, 2023-02-10:
See https://drake.mit.edu/pydrake/pydrake.geometry.html#pydrake.geometry.Meshcat.GetGamepad.

> Note that in javascript, gamepads are not detected until users “opt-in” by pressing a gamepad button or moving the thumbstick in the Meshcat window. If no gamepad information is available in javascript, then no messages are sent and the returned gamepad index will not have a value.

I didn't see you open Meshcat in a browser and start using the gamepad.

The `No gamepad found.` output looks like it's coming from your code, not Drake.

adeeb10abbas, 2023-02-11:
You're absolutely correct, it's just hitting this bit of the code - 
```
    gamepad = meshcat.GetGamepad()
    if gamepad.index != None: # use gamepad teleop
        print("Gamepad found.  Using gamepad teleop.")
        gamepad_command = builder.AddSystem(GamepadCommand(meshcat))
        builder.Connect(gamepad_command.get_output_port(),
                        controller.get_input_port(0))

    else: # use keyboard teleop
```
When I start using it nothing happens. I can post the video but it's just a static simulation with nothing happening.

jwnimmer-tri, 2023-02-11:
Next step, check your browser console for any errors or warnings:

https://github.com/RobotLocomotion/drake/blob/153195c80cc35ca7dfee91dca3c7e76238708434/geometry/meshcat.html#L41-L51

adeeb10abbas, 2023-02-11:
Yep checked that as well. I don't think I see anything relevant but could be wrong. This is what I see - 
[image]

jwnimmer-tri, 2023-02-11:
> Note that in javascript, gamepads are not detected until users “opt-in” by pressing a gamepad button or moving the thumbstick in the Meshcat window. If no gamepad information is available in javascript, then no messages are sent and the returned gamepad index will not have a value.

This is a _per session_ opt-in.

Given that the call to `meshcat.GetGamepad()` happens almost immediately after starting the program, I had a difficult time believing that you refresh the browser window, grab focus, and start pushing gamepad buttons before the code reaches the if-else branch that gives up on the gamepad forevermore.

I think you need to pause your script (maybe with a Meshcat button like "Begin...") to allow the user to opt-in their gamepad before you decide which systems to create.

Alternatively, you could add _both_ keyboard and gamepad systems (plumbed through an Adder) so that both can take effect at once.

jwnimmer-tri, 2023-02-11:
On the other hand, if you're saying that you have some _other_ gamepad that works, but the PS5 specifically does not, then we should look more carefully at your "Console" window (not "Issues").

adeeb10abbas, 2023-02-11:
I see what you mean. My bad I entirely misunderstood the per-session opt-in bit. I will give the suggestions a shot tomorrow! 

My console window was entirely empty. I just shared the only "issue" I saw :) And no, I don't have any other gamepads that work with this. 

Thanks a lot for helping out!

ggould-tri, 2023-02-13:
Assigned geometry illustration (for lack of a clear better component) and jwnimmer-tri (since he is already on this) for now.  Feel free to change these up.

jwnimmer-tri, 2023-02-13:
SGTM.

I think one conclusion here is that we need better documentation (and demos?) for how to set up and use gamepads.

adeeb10abbas, 2023-02-13:
Yes, it worked when I added some delay 😅 I agree better documentation would have helped this. Thanks for the help. I don't mind recording demos if you'd like that to address this :)

RussTedrake, 2023-02-16:
Does that mean we should leave this open until the documentation is better, or should we close the issue?

jwnimmer-tri, 2023-02-16:
I think leave it open.  I've renamed it now.