RobotLocomotion/drake #18768: LeafSystem::DeclarAbstractOutputPort argument is spelled `alloc` in pydrake, when c++ uses `alloc_callback`.

RussTedrake, 2023-02-12:
This one is trivial.  But the python binding doesn't match the c++ code. (There might be other offenders nearby, too). We should tidy that up.

mitiguy, 2024-05-28:
In leaf_system., it seems the relevant function definition is:

```
  LeafOutputPort<T>& DeclareAbstractOutputPort(
      std::variant<std::string, UseDefaultName> name,
      typename LeafOutputPort<T>::AllocCallback alloc_function,
      typename LeafOutputPort<T>::CalcCallback calc_function,
      std::set<DependencyTicket> prerequisites_of_calc = {all_sources_ticket()});
```

As of now, the file framework_py_systems.cc uses the arguments
py::arg("name"), py::arg("alloc"),  py::arg("calc"), py::arg("prerequisites_of_calc")
so the mismatches are  "alloc" (Python)  vs.  alloc_function (C++)
and  "calc" (Python)  vs. "calc_function" (C++).

The title of this issue may be outdated as it mentions an argument spelled ```alloc``` in pydrake, when C++ uses ```alloc_callback```.  I could not find a use of ```alloc_callback``` in Drake.

jwnimmer-tri, 2024-05-29:
I think it was just a typo in the original issue.  The LeafSystem C++ name has been `alloc_function` since its inception.

jwnimmer-tri, 2024-05-31:
Fixed in #21478.

closed 2024-05-31 by jwnimmer-tri