RobotLocomotion/drake #18775: CI Resource Allocation Issues with Gurobi Tests

svenevs, 2023-02-13:
With #18663 running we are seeing some tests that may need to request more resources in their `BUILD.bazel` file.  Plan for this ticket: monitor these builds and if failures continue find a fix appropriate for each test.

### global_inverse_kinematics_reachable_test

https://github.com/RobotLocomotion/drake/blob/da207dbf1a8ec7156388515e7aba6ca8999aca53/multibody/inverse_kinematics/BUILD.bazel#L382-L396

[Discussion on slack](https://drakedevelopers.slack.com/archives/C270MN28G/p1676139844632869), failures:

```
[4:12:01 AM]  ==================== Test output for //multibody/inverse_kinematics:global_inverse_kinematics_reachable_test:
[4:12:01 AM]  Using drake_cc_googletest_main.cc
[4:12:01 AM]  
[4:12:01 AM]  [==========] Running 1 test from 1 test suite.
[4:12:01 AM]  [----------] Global test environment set-up.
[4:12:01 AM]  [----------] 1 test from KukaTest
[4:12:01 AM]  [ RUN      ] KukaTest.ReachableTest
[4:12:01 AM]  Set parameter ServerPassword
[4:12:01 AM]  Set parameter TokenServer to value "172.31.29.246"
[4:12:01 AM]  Warning for adding constraints: zero or small (< 1e-13) coefficients, ignored
[4:12:01 AM]  Warning for adding constraints: zero or small (< 1e-13) coefficients, ignored
[4:12:01 AM]  Warning for adding constraints: zero or small (< 1e-13) coefficients, ignored
[4:12:01 AM]  -- Test timed out at 2023-02-11 09:12:00 UTC --
```

- https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-everything-release/988/
- https://drake-jenkins.csail.mit.edu/job/linux-focal-unprovisioned-clang-bazel-nightly-everything-debug/669/

### multibody/benchmarking:homecart_global_ik_test

https://github.com/RobotLocomotion/drake/blob/da207dbf1a8ec7156388515e7aba6ca8999aca53/multibody/benchmarking/BUILD.bazel#L81-L99

[Discussion on slack](https://drakedevelopers.slack.com/archives/C270MN28G/p1676301017489389), failures:

```
[3:34:58 AM]  ==================== Test output for //multibody/benchmarking:homecart_global_ik_test:
[3:34:58 AM]  ==280260== Memcheck, a memory error detector
[3:34:58 AM]  ==280260== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
[3:34:58 AM]  ==280260== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
[3:34:58 AM]  ==280260== Command: multibody/benchmarking/homecart_global_ik_test --benchmark_min_time=0
[3:34:58 AM]  ==280260== 
[3:34:58 AM]  2023-02-13T08:10:07+00:00
[3:34:58 AM]  Running multibody/benchmarking/homecart_global_ik_test
[3:34:58 AM]  Run on (16 X 3399.92 MHz CPU s)
[3:34:58 AM]  CPU Caches:
[3:34:58 AM]    L1 Data 32 KiB (x8)
[3:34:58 AM]    L1 Instruction 32 KiB (x8)
[3:34:58 AM]    L2 Unified 1024 KiB (x8)
[3:34:58 AM]    L3 Unified 25344 KiB (x1)
[3:34:58 AM]  Load Average: 15.32, 14.46, 14.19
[3:34:58 AM]  Set parameter ServerPassword
[3:34:58 AM]  Set parameter TokenServer to value "172.31.29.246"
[3:34:58 AM]  Warning for adding constraints: zero or small (< 1e-13) coefficients, ignored
[3:34:58 AM]  Set parameter ServerPassword
[3:34:58 AM]  Set parameter TokenServer to value "172.31.29.246"
[3:34:58 AM]  Warning for adding constraints: zero or small (< 1e-13) coefficients, ignored
[3:34:58 AM]  -- Test timed out at 2023-02-13 08:34:58 UTC --
[3:34:58 AM]  ========INFO: From Testing //multibody/benchmarking:homecart_global_ik_test:
```

- https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-everything-valgrind-memcheck/372/consoleFull
- https://drake-jenkins.csail.mit.edu/view/Weekly%20Production/job/linux-jammy-clang-bazel-weekly-gurobi-valgrind-memcheck/31/consoleFull

BetsyMcPhail, 2023-02-14:
Other related timeouts:

- `//multibody/inverse_kinematics:global_inverse_kinematics_reachable_test` timed out in [linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-unprovisioned-gcc-bazel-nightly-everything-debug/219/)

-  `//solvers:mixed_integer_rotation_constraint_limit_test` timed out in [linux-focal-clang-bazel-continuous-everything-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-everything-leak-sanitizer/1610/).

jwnimmer-tri, 2023-02-15:
Transcribing my proposal from slack:

```patch
--- a/solvers/gurobi_solver.cc
+++ b/solvers/gurobi_solver.cc
@@ -1257,6 +1257,17 @@ void GurobiSolver::DoSolve(
                      static_cast<int>(merged_options.get_print_to_console()));
   }
 
+  // Default the option for number of threads to respect the OMP environment
+  // variable (but only if the user hasn't set the option already), and only if the OMP limit is sufficiently small.
+  if (merged_options.GetOptionsInt(id()).count("Threads") == 0) {
+    if (char* omp_num_threads_str = std::getenv("OMP_NUM_THREADS")) {
+      const int max_num_threads = ParseInt(omp_num_threads_str);
+      // https://www.gurobi.com/documentation/9.5/refman/threads.html
+      if (max_num_threads < 32)
+        merged_options.SetOption(id(), "Threads", max_num_threads);
+    }
+  }
+
   for (const auto& it : merged_options.GetOptionsDouble(id())) {
     if (!error) {
       SetOptionOrThrow(model_env, it.first, it.second);
```

jwnimmer-tri, 2023-02-20:
@svenevs we're ready now to try the drake-ci patch again to enable non-exclusive Gurobi tests.

svenevs, 2023-02-20:
=> https://github.com/RobotLocomotion/drake-ci/pull/202

BetsyMcPhail, 2023-02-21:
The CI change has been re-applied.

BetsyMcPhail, 2023-02-22:
No CI failures have appeared so I think we can call this one closed.