RobotLocomotion/drake #18787: Clang compiler unable to link against fortran

SeanCurtis-TRI, 2023-02-14:
### What happened?

Simply executing:

    bazel build --config=clang //geometry:meshcat_manual_test

produces the error message:

```
ERROR: /home/seancurtis/code/drake/geometry/BUILD.bazel:481:16: Linking geometry/meshcat_manual_test failed: (Exit 1): clang failed: error executing command (from target //geometry:meshcat_manual_test) /usr/lib/llvm-12/bin/clang @bazel-out/k8-opt/bin/geometry/meshcat_manual_test-2.params

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
/usr/bin/ld.gold: error: cannot find -lgfortran
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Target //geometry:meshcat_manual_test failed to build
```

However, defaulting to gcc (omitting `--config=clang`) compiles fine. There are no extraneous requests for snopt in any of my local (workspace or user account) bazelrc files.

It has been found that commenting out [this line](https://github.com/RobotLocomotion/drake/pull/15892/files#diff-148d056c20dd1e49668baccad3008ebd63f4e80195df663c6c76a4c9271cd958) alleviates the problem locally (obviously leaving further problems lurking for creating wheels).

### Version

5a5b98c6025994330caf097560da5d3a06b720fe

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2023-02-14:
Note to self: check why CI isn't finding this.

jwnimmer-tri, 2023-02-21:
Interesting; lapack _does_ actually load gfortran.

```console
jwnimmer@call-cps:~/jwnimmer-tri/drake$ readelf -d /usr/lib/x86_64-linux-gnu/liblapack.so | grep fort
 0x0000000000000001 (NEEDED)             Shared library: [libgfortran.so.5]
jwnimmer@call-cps:~/jwnimmer-tri/drake$ ldd /usr/lib/x86_64-linux-gnu/liblapack.so | grep fort
	libgfortran.so.5 => /lib/x86_64-linux-gnu/libgfortran.so.5 (0x00007f6ee20c6000)
```

jwnimmer-tri, 2023-02-21:
Very weird.  CI is happy:
https://drake-jenkins.csail.mit.edu/view/Experimental/job/linux-jammy-clang-bazel-experimental-release/10/console

jwnimmer-tri, 2023-02-22:
Aha.  I think its another flavor of #18669.  The nvidia drivers on our workstations end up installing gcc 12, but not the gfortran 12 developement files.

`sudo apt install `libgfortran-12-dev` fixes it.

I'll try to figure out how to automate that.

PR #18925 (2023-03-02): [setup] Automatically fix more GCC 12 installation mistakes
Thanks, NVidia.

Closes #18787.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18925)
closed 2023-03-02 by jwnimmer-tri