RobotLocomotion/drake #18790: Request for AddDistanceFromLineConstraint for Drake IK

zevbo, 2023-02-15:
**Is your feature request related to a problem? Please describe.**
I am currently working on an IK problem where I would like to add a constraint that stops the center of a frame from being in a certain cylinder. The only way to do it with the current IK that I can think is say that the point must be farther than the radius of the cylinder away from a bunch of points on the axis of the cylinder.

**Describe the solution you'd like**
 I would love an `AddDistanceFromLineConstraint`, making the line the axis of the cylinder, and the distance the radius of the cylinder. 

**Describe alternatives you've considered**
Alternatively, `AddExcludeFromPolyhedronConstraint` would also be sufficient as I could approximate the cylinder with a number of surfaces. That said, I imagine that would be much harder to throw a solver at for convexity reasons.

jwnimmer-tri, 2023-02-15:
@hongkai-dai would a [DistanceConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_distance_constraint.html) work for this use case?  One geometry could be the Cylinder, and the other a tiny sphere on the frame.

hongkai-dai, 2023-02-15:
> would a [DistanceConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_distance_constraint.html) work for this use case? One geometry could be the Cylinder, and the other a tiny sphere on the frame.

@jwnimmer-tri , yes I also thought about that. Hence I asked the question if the user wants a distance to a infinite length line or a line segment. If it is a line segment, then we can use DistanceConstraint between a sphere and a cylinder.

But I still think it is valuable to provide a special constraint just for the point-to-line distance case. The reason is that if we use the DistanceConstraint between a sphere and a cylinder, the user will need to add a collision geometry of a sphere and a cylinder, and maybe also need to exclude collision between this specific sphere (and the cylinder) to any other collision geometries in the scene, if there are other distance constraints (like MinimumDistanceConstraint) in the IK problem. Hence for the users, it might be more effort to use DistanceConstraint with the sphere and the cylinder, compared to creating a specific PointToLineDistanceConstraint. This was the same reason why I created [PointToPointDistanceConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_point_to_point_distance_constraint.html) . What do you think?

zevbo, 2023-02-17:
I suppose I could do it with the `DistanceConstraint`; I had not really thought about that. But I do agree with @hongkai-dai  that its much harder to think abut creating geometries that don't correspond to a real world object for one IK problem, than it is to have a simple designated constraint. I definitely appreciate the convenience of the [PointToPointDistanceConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_point_to_point_distance_constraint.html), so I do ultimately agree that `PointToLineDistanceConstraint` would be useful.

closed 2023-02-22 by hongkai-dai

hongkai-dai, 2023-02-22:
@zevbo we add PointToLineDistanceConstraint into Drake master through #18817 ,  let me know if that does not solve your problem.

PR #18817 (2023-02-22): [Inverse Kinematics] Add PointToLineDistanceConstraint.
Constrain the distance between a point and a line to be within certain range.

Resolves #18790

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18817)