RobotLocomotion/drake #18791: Consider macOS support for two most recent Python versions

jwnimmer-tri, 2023-02-15:
Homebrew's rolling transitions from one Python to the next one are a bit brutal on our users (#18262).

It might help smooth over that process if we supported two possible Python versions on macOS at once, instead of a flag-day transition.

Possibly #18719 would be a sufficient answer for from-source users, and then for binary users we could publish two wheels (e.g., cp310 and cp311).

jwnimmer-tri, 2023-02-23:
The plan here is probably that I should finish #18719 first.

After that, we should look into running two macOS wheel builds in CI.  The wheel build can use the same infra as #18719 to switch which Python it's using.

Possibly in that case, the wheel build would need to use more things from pip and fewer from brew (ala #8392).

RussTedrake, 2023-03-01:
💯  from me.

torch doesn't support python3.11 on mac yet.  torch 2.0 adds that support, but it is not available on pypl yet.  I was able to get my manipulation repo working on mac using the torch nightlies, but after a few weeks of occasionally bashing my head against trying to get my mac CI to work with the torch nightly wheels, I've decided to just disable mac CI for my manipulation repo until torch2.0 lands on pypl.   It's been an exercise that I wouldn't wish upon my enemies, and certainly not my friends!

jwnimmer-tri, 2023-03-01:
To your manipulation repo in particular, it seems you're using the `drake_bazel_installed` as a source for Drake.  I think in this ticket we'd offer at most either `drake_cmake_external` (to build from source) or `PyPI` (i.e., using a wheel) for the off-nominal python version.  So I think one condition here for you would be switching your repo to use Drake from PyPI.

jwnimmer-tri, 2023-04-20:
For the record, https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_external should probably work for most Python versions now (assuming the user has installed sufficient pip dependencies) as of #19023.

This ticket will track "Ship multiple macOS wheels to PyPI" feature request.

jwnimmer-tri, 2023-10-05:
I renamed this to _Build nightly macOS wheels for two Python versions_.

The new objective is that https://drake.mit.edu/pip.html#nightly-releases provides wheels for the two most recent Homebrew versions of Homebrew Python (which as of today would be 3.11 and 3.12).

Nothing will change for our stable releases.  The "download release" scripts will only fetch the _stable_ (i.e., drake-default) version of Python on macOS ([currently 3.11](https://drake.mit.edu/installation.html)).

I don't mind one way or another whether the Staging builds compile both wheels and we throw one away, or if it just compiles the one stable wheel we'll ship.

jwnimmer-tri, 2023-10-05:
Per f2f: @mwoehlke-kitware and I seem to agree that we should try to have just one Mac Jenkins Wheel job (per architecture) for both Python versions (as opposed to 1 job per version).  That way, it can share the local bazel disk cache so the second build (for 3.12) should be a lot faster than the first (for 3.11).  The important thing will be to `rm -rf` things like the _install_ tree between the two builds.  We shouldn't remove the local disk cache inbetween.

mwoehlke-kitware, 2023-11-28:
Do we believe it's possible to build a 3.12 wheel today? How would one go about doing that? (I'm guessing a bunch of stuff needs to be installed via `brew`?)

mwoehlke-kitware, 2023-11-28:
AFAICT, we aren't installing Python packages. The existing CI must rely on the dependencies already being installed?

It looks like the wheel builder is going to have to change to build Drake in a venv. The usual user setup step, `pip install requirements.txt`, refuses to execute:
```
$ pip3.12 install -r setup/mac/binary_distribution/requirements.txt
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install.
```

jwnimmer-tri, 2023-11-28:
I imagine it's easy to manually put a valid `python3.12-config` symlink into the venv.

PR #20616 (2024-01-18): Build macOS wheels for multiple Python versions
Rework macOS wheel build scripts to use Python-version-specific permanent locations for all operations, with temporary symlinks used for operations shared with the Linux wheel builds that expect non-versioned locations. Also rework builder to support a set of Python targets and to be able to build multiple wheels in a manner similar to how things work for Linux.

This changes the macOS wheel build to produce wheels for (as of writing) both Python 3.11 and 3.12.

Fixes #18791.
Fixes #19098.

- - -
This change is [[image]](https://reviewable.io/rev [... cut]
closed 2024-01-18 by jwnimmer-tri