RobotLocomotion/drake #18792: Clarify pip artifact url availability

svenevs, 2023-02-15:
Particularly where macOS is concerned, our [current docs](https://drake.mit.edu/pip.html) give the user the impression that they'd be able to snag a `cp311-cp311` macOS nightly wheel from 2023-02-01, but they cannot -- those were `cp310-cp310` until we switched on 2023-02-15 to python 3.11.

We should concoct a description that clearly explains this, and possibly just start manually denoting the macOS transition dates (as well as when a linux was added, e.g., 311)?

There are ways to [get a listing of objects from s3 for drake-packages](https://stackoverflow.com/a/15633672), but last time we looked into this it was too difficult to get.  From what I can see, though, that would potentially increase our s3 costs quite a bit (we get billed for querying).  AKA there isn't a good way to do this automatically that I can see.

CC @RussTedrake @jwnimmer-tri @BetsyMcPhail via [original slack discussion](https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1676460411555109).

RussTedrake, 2023-02-15:
Thanks.  The second (closely related) idea from the slack channel was to add a sentence to [the pip page](https://drake.mit.edu/pip.html) which just explains what the cp310-cp310 (etc) incantation means.  Not everyone knows!

svenevs, 2023-02-15:
> explains what the cp310-cp310 (etc) incantation means

Absolutely, we should add that.  The first one is the [python tag](https://peps.python.org/pep-0425/#python-tag), the second the abi tag (sometimes that can be none, but for our stuff it is not).  The `cp` means `CPython`, the only distribution we support and package for (rather than pypy, ironpython, jython, etc), and `310` is `{python major}{python minor}` version numbering.

> One idea -- a nightly (really, mid-day) Jenkins job that uses python3-boto to list the nightly folder, and then uploads an index.html file that we could explicitly link to from the website.

This could certainly be done.  xref #17551 for developer convenience, I just copied the `index.html` to the `drake/nightly` as a test: https://drake-packages.csail.mit.edu/drake/nightly/index.html the other AWS settings are already configured for static website hosting.  I think the only additional step needed is to set the headers up to not cache in the browser at all or do so for only a few hours (on upload, see also https://github.com/RobotLocomotion/drake-ci/blob/e0f93e61ea00b6419ede0fb71f7c986ed3dfeb71/tools/upload-to-aws.py#L115).

We could populate and upload a new `index.html` automatically from CI.  Would we want to just list them out by date?

```html
<h2>2023-02-15</h2>
<ul>
  <li>artifact 1</li>
  <li>...</li>
  <li> artfiact N</li>
</ul>

<h2>2023-02-14</h2>
...
```

Or have separate sections, say, "mac wheels", "linux wheels", "mac .tar.gz", "linux {codename} .tar.gz"?  I don't have a strong preference.  I think by date is easiest, and the theory here being that if somebody does `ctrl+F` and searches for `YYYYMMDD` then they will see all available artifacts for that date immediately, and can select the desired artifact.  Then `pip.md` would expand its verbiage and explain search tactics, making a note of `cp310` vs `cp311` things that can come up with examples.

jwnimmer-tri, 2023-02-15:
My imagination for the `index.html` was literally a (sorted) directory listing, with links, nothing semantically fancy at all.

It might be worth finding out what `pip` supports.  I know it can use directory-listing index files to install nightlies from.  We'd probably want to be compatible with that.

```
pip3 install --pre drake --index-url https://drake-packages.csail.mit.edu/drake/nightly/index.html
```

svenevs, 2023-02-15:
Relevant links for further investigation:

- pip docs: https://pip.pypa.io/en/stable/development/architecture/package-finding/
- indexu url standard: https://peps.python.org/pep-0503/
- examples:
    - https://pypi.org/simple/
    - https://pypi.org/simple/drake/
    - view-source:https://pypi.org/simple/drake/

sherm1, 2023-02-16:
Assigned to Jeremy for disposition.

RussTedrake, 2023-05-30:
N.B. -- having the `index.html` rollup that would enable `pip install` from nightly (on all platforms) would help my downstream repo management considerably.

jwnimmer-tri, 2023-06-15:
See also #19130 related to gatekeeper.  We should check if Gatekeeper complains if the user clicks on the `.whl` hyperlink to download it using Safari and then does `env/bin/pip install my_localfile_drake.whl`.  If it complains, then we should likewise replace demote `whl` hyperlinks (at least for macOS) to be curl commands to copy-paste instead.

jwnimmer-tri, 2023-07-27:
The victory condition is to make it sufficiently easy for users to install pydrake nightlies with pip.

- Using a daily cron job, refresh the pip `index.html` page (pushing the new copy to S3), based on all available nightly wheels (the ones that haven't been reaped yet, e.g., ~50-some days worth, on all platforms).
- Fix the instructions to use the index html instead of hard-coded URLs.
  - To install nightly, it would be `drake < 1`.
  - To install specific, it would be `drake == 0.0.YYYYMMDD`.
- Be careful with Glacier.  Our index.html generator must not list those.
  - Ideally, the http server would 404 if the file is in Glacier.  Check to see if we can automatically prevent it from resurrecting.

svenevs, 2023-09-11:
I've updated the description of the issue to reflect the updated status.  Nightly wheel builds should begin updating the generated `index.html` component (https://github.com/RobotLocomotion/drake-ci/pull/241), and we'll be able to monitor the s3 rules this week.

Where documentation is concerned, it's not clear to me if what I've written on the updated description breaking down how / why `pip install --extra-index-url https://drake-packages.csail.mit.edu/whl/nightly/ 'drake<0.1'` works (including why that version number specifically exists).  But it should at least help in writing those docs.

I'll try and track down what command ended up downloading too many things (we should do that before advertising this / landing the drake docs PR) next.

jwnimmer-tri, 2023-09-12:
The CI job seems have worked properly last night.  What remains here is to update our documentation to reflect the updated `pip install` procedure to use nightly builds.

jwnimmer-tri, 2023-09-12:
For the install docs, I think the primary use case we should focus on is a pinned Drake.  So `drake==0.0.YYYYMMDD`.  Once that has been explained, we can also mention that to fetch "latest" you can use `drake<0.1`.

jwnimmer-tri, 2023-09-14:
>  See if we can reproduce the thing jeremy said that it downloaded ALL wheels using a fairly unconstrained "drake" pip install.

I think the risk-reward is in favor of deploying the documentation and advertising this feature to our users, even before we've finished this step.  The only downside is a little extra $$ until we hunt down the problem, and that's not a good enough reason to delay its rollout.

svenevs, 2023-09-14:
@jwnimmer-tri what version of pip were you using?

https://github.com/pypa/pip/issues/9284

If it was that version then I think we can consider it resolved.

jwnimmer-tri, 2023-09-14:
It was mostly likely the Jammy default of `22.0.2`.  The only other thing it could have been was the latest (`venv/bin/pip install --upgrade pip`) but I don't usually do that in my venvs anymore.

svenevs, 2023-09-19:
All components complete here!

closed 2023-09-19 by svenevs