RobotLocomotion/drake #18803: Parse closed kinematic chains from SDF (and perhaps also URDF)

RussTedrake, 2023-02-16:
SAP now supports constraints, but currently users can only define those constraints programmatically.

Closed kinematic chains are a common feature in robots (e.g. Cassie, or in many underactuated hands). We should support parsing a robot with a closed kinematic chain directly from SDF and/or URDF.  The specification in URDF is such that one cannot define a closed chain without additional tags, but in SDF it is possible to define a closed chain just by adding the already defined links and joints.  Presumably we through an error in this case, but I have not verified that.

We used to have a Drake-specific "loop_joint" tag.  You can still find vestiges of it in our codebase; the URDF parser even reports `    Error(*node, "loop joints are not supported in MultibodyPlant");` if it encounters one.  I believe @sherm1 feels that it's better for the user to be able to decide where the mechanism should be split and closed with a constraint (vs the true joints), and that it's better to split in the middle of a link and then have a constraint which welds the link back together (vs a lower-DOF constraint implemented at a joint).  I am ok with requiring a custom tag.

A victory condition here would be to have the `four_bar.sdf` and even `strandebeest` use constraints without having to add them programmatically.  They should be specified in the robot description file.

sherm1, 2023-02-16:
FYI I opened a related draft PR #18804 that can turn these into spanning tree + constraint models.

rpoyner-tri, 2023-02-16:
There are potentially two pieces of work here. (1) make the parser map closed chains onto appropriate MbP features; (2) make MbP support closed chains at all. Do we believe enough of (2) is done to undertake (1)? Based on Sherm's comment above and other discussion, I'm not sure.

jwnimmer-tri, 2023-02-16:
My intuition is that we could write a `four_bar.sdf` or `four_bar.urdf` file that uses a to-be-added custom tag to define the constraint, at which point MbP already offers everything we need.  It's probably worth trying to write down that exemplar input file as the next step here.

In that design, the user would still need to point their joints in right direction by hand in the xml, and annotate the constraint by hand in xml, but it would meet the victory condition of not calling MbP API to set up the model.  (Other than to put it into SAP mode, of course.)

sherm1, 2023-02-16:
There are definitely two different problems, greatly differing in difficulty:
1. Allow users _some_ way to construct tree+constraint models, or
2. Read unmodified sdf files containing only links and joints and automatically turn those into internal tree+constraint systems

In general the second case is _much_ harder. We probably want to provide option 1 first, if only to make sure we have suitable underlying functionality. If we want to do option 2, we need some code that can produce a reasonable tree+constraint model, which can be quite difficult. Just for a taste: massless bodies (which are quite reasonable in linkages) must never be placed as leaf nodes in the tree. #18804 solves the general problem.

sherm1, 2023-02-16:
BTW for the automated (option 2) all we need is Weld constraints. That's all users need too but they might expect a large collection of "loop joint" constraints instead. That leads to asymmetric behavior where some joints behave differently than others (e.g. there are not q's and v's in the state corresponding to the loop joints). The method I advocate for mapping loops to trees is:
- for each kinematic loop in the system (there can be many), choose one body
- cut that body in half, breaking the loop (each body gets half the mass properties)
- weld the two halves together with a Weld constraint
- assemble and run.

With that approach the joints are all treated identically.

jwnimmer-tri, 2023-02-16:
It sounds like you have a lot of experience here @sherm1.  Could we convince you to write a `four_bar.sdf` or `four_bar.urdf` sample file that demonstrates that approach?

sherm1, 2023-02-16:
Sure -- do you mean the option 1 approach where we postulate the existence of a `<drake::weld_constraint>` tag or something like that? The option 2 sdf file leaves all the fancy stuff hidden; I'll do that one for contrast.

jwnimmer-tri, 2023-02-16:
Yes, exactly option 1.  (Doing both options is fine, too.)

The goal would be to hand that file to a software developer and say "add parsing code so that a unit which loads this file passes".

sherm1, 2023-02-20:
See #18829 for possible sdf descriptions

RussTedrake, 2023-03-19:
related to https://github.com/RobotLocomotion/drake/issues/18803

jwnimmer-tri, 2023-03-19:
FYI there's some type in your last; it's a self-link back to this issue.

RussTedrake, 2023-03-19:
oops.  updated now.  thanks.

jwnimmer-tri, 2023-04-20:
@rpoyner-tri If you're looking for the next suggestion for a parsing project (after 18773), I'll upvote this one.

sherm1, 2023-04-20:
FYI I'm working on the graph processing that can handle this as we speak. If the parser can get the links, joints, and constraints into MbP we'll be able to model correctly.

rpoyner-tri, 2023-09-21:
Noting #20232 here, for support of closed chains from MJCF as well.

sherm1, 2023-09-21:
Current graph processing draft is #20225

jwnimmer-tri, 2024-02-19:
Having just re-read all of the posts, I'll summarize here for the next person who wanders by:

The call to action here is to write the SDFormat custom parsing logic (and syntax documentation) that can parse [four_bar_weld.sdf](https://github.com/RobotLocomotion/drake/blob/master/examples/multibody/four_bar/dev/four_bar_weld.sdf) using the custom `weld_constraint` tag:

```xml
    <drake:weld_constraint>
      <parent>Ccm</parent>
      <child>Scm</child>
    </drake:weld_constraint>
```

RussTedrake, 2024-09-29:
FWIW (since I didn't know until I stumbled on the code), https://github.com/RobotLocomotion/drake/pull/19780 introduced `<drake:ball_constraint>` for URDF and SDF, which does *enable* authoring loop joints, though not in the way that @sherm1 prefers (by sawing a bone in half and welding it back together).

sherm1, 2024-09-30:
BTW breaking a loop at a ball joint is the one case where using a constraint-based loop joint is arguably harmless. We delete 3 differential equations and add 3 constraint equations. Although we lose the internal coordinates q and v, those usually aren't much use for a ball joint anyway, especially when modeling with a quaternion. And the alternative is to retain the 3 DEs, cut a link, and add 6 constraints to glue it back together.

sherm1, 2026-09-24:
Now that we have a draft of automatic closed-loop handling in Drake it is time to revisit this. Despite the optimistic comments above, SDFormat _cannot_ express an unassembled mechanism, which is the only practical way to describe closed-topology systems. I propose to extend the `<drake:parent>` and `<drake:child>` tags (currently allowed under `<drake:joint>`) to be accepted in `<joint>` as well. The idea would be to use those to provide frames which would be passed directly to Drake's AddJoint() and thus provide Drake semantics. An example for a revolute joint would be:
```
<joint name="driver_coupler" type="revolute">
  <!-- Still required by SDFormat for topology validation. -->
  <parent>driver</parent>
  <child>coupler</child>

  <!-- Exact Drake joint frames. -->
  <drake:parent>Dc</drake:parent>
  <drake:child>Cd</drake:child>

  <axis>
    <xyz>0 -1 0</xyz>
  </axis>
</joint>
```
This would get us one step closer to making the current "internal use only" feature a fully supported Drake capability. Other things we need to resolve are detailed in #24908, and we need to provide an Assemble() analysis type along the lines implemented already in the four_bar_auto example.

WDYT?

jwnimmer-tri, 2026-09-24:
Why do you say that "unassembled mechanism ... is the only practical way to describe closed-topology systems"? I haven't actually looked, but I have been assuming that existing robot descriptions in the wild with closed chains already have their default posture be correctly assembled. Is that not the case? Do you have some examples you could link to?

Or, even if SDFormat doesn't support unassembled, do URDF or MJCF support it?  It would be unusual for simulation authors to want to use SDFormat for the robot, and if we do need to enhance a custom Drake parser, I don't think SDFormat should be the first spot we touch. Most robots are either URDF or MJCF (or USD, perhaps imported via URDF).

sherm1, 2026-09-24:
To input an assembled closed-topology mechanism, one must first perform an assembly analysis which is a nasty nonlinear problem resulting in configurations that must be input as 15-digit numbers. To perform that analysis first requires a model of the system and its constraints expressed more conveniently; currently there is no Drake file format that permits that. None of sdf, urdf, or mjcf can express an unassembled system. usd can but we don't support it.

The only practical way non-usd mechanisms can be expressed is to chop them into tree + constraint manually and input that directly. Then the tree can be defined in a nice configuration. We're trying to allow for mechanisms that are input without being manually chopped; that requires either using the Drake API, supporting USD, or adding extensions somewhere. I'm not sure whether urdf could be extended the same way but could investigate.

sherm1, 2026-09-24:
DR Legs is the most interesting closed-topology robot examples I've seen:

[image]

This is from Disney Research's Kamino project. It's defined by a USD file [here](https://github.com/newton-physics/newton-assets/blob/main/disneyresearch/dr_legs/usd/dr_legs_with_meshes_and_boxes.usda). The loops are not manually broken, although it does appear the USD description is already assembled (I've got an email to the developer to see how they did that -- I'm guessing this was exported from a CAD package where the assembly was done).

jwnimmer-tri, 2026-09-24:
Yes. My hope is that robots typically come from CAD, and CAD can typically export an already-assembled robot. In which case, the simulation / control developer does not need to use Drake to assemble a robot, rather they can just export it in assembled form already (to 15 places).

In other words, it is not clear to me that "perform an assembly analysis" is in scope for the capabilities requested / necessary here.

I could see the possible need for taking an existing historical model which is expressed with an equality constraint (in Drake or MuJoCo) or bushing (in Drake) and outputting a replacement model that uses an assembled closed loop instead of the constraint. I imagine, however, that we could do that without introducing special joint definition, rather by just reading out the kinematics of a satisfactory constrained posture.

sherm1, 2026-09-28:
It may be the case that real world robots all come from CAD, but that is not the case for our internal examples and test cases (I'm thinking of Strandbeest, DR Legs, and my current clumsy pedagogical examples). It would be very handy to have _some_ file format where we can build these models by hand.

BTW practical need for assembly does arise in design problems. Even if the original of a mechanism was assembled, it won't be if any of the links change size or attachment frames move as a result of parameterization. In those cases the workflow is: tweak design, reassemble, simulate, evaluate objective.

I'd like to add `drake:parent` and `drake:child` to our sdf parser. I guess we don't need to announce those in release notes if they aren't useful externally but I'd like to have them available internally (the whole auto loop breaking utility is internal for now).

sherm1, 2026-10-22:
@SeanCurtis-TRI  had a better idea for the proposed SDFormat extension. We should just provide an attribute that makes `<joint>` treat frames the way `<drake:joint>` does already. An empty element tag appears to be the best way to do this rather than a literal Xml attribute (similar to other Drake extension like `drake:declare_convex`). So the above joint would be specified like this:
```xml
<joint name="driver_coupler" type="revolute">
  <drake:independent_frames/>
  <parent>Dc</parent>
  <child>Cd</child>

  <axis>
    <xyz>0 -1 0</xyz>
  </axis>
</joint>
```

Thoughts?

SeanCurtis-TRI, 2026-10-22:
Why is the empty tag better than an attribute?

sherm1, 2026-10-22:
There are a number of awkward features of the Xml attribute idea. First is that they have to have an argument:
`<joint name="coupler_rocker" type="revolute" drake:independent_frames="true">`

Then error checks are required to make sure the meaningless argument isn't, say, "false" and that a typo like just `drake:independent_frames` isn't silently ignored (which would happen now).

From Claude: Attribute vs. element. An older Drake silently ignores an unknown _attribute_, but errors on an unknown drake: _element_. A tree model using the attribute would therefore load in older Drake as a different mechanism (loop models fail at Finalize() anyway). If that matters, a child element <drake:independent_frames/> gets that rejection for free. An element also removes the two drawbacks of the attribute:
- Misspellings are caught. An unknown drake: element is already an error through CheckSupportedElements, so <drake:independant_frames/> fails to load. The new attribute check drops out of the plan; the name just goes into supported_joint_elements.
- Older Drake rejects the file. It reports "Unsupported SDFormat element in joint", rather than quietly building a different mechanism.

SeanCurtis-TRI, 2026-10-22:
Element sounds fine.