RobotLocomotion/drake #18807: ModuleNotFoundError: No module named 'pydrake.planning'

ZZHPKU, 2023-02-17:
### What happened?

I installed pydrake by apt under Ubuntu 20.04. When I run code

"from pydrake.planning import SceneGraphCollisionChecker"

I met 

"ModuleNotFoundError: No module named 'pydrake.planning'".

I can run other pydrake package. It looks like pydrake.planning is missing. 

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

apt install drake

jwnimmer-tri, 2023-02-17:
What is the output of `dpkg -l drake-dev`?  (In other words, which version of Drake are you using?)

jwnimmer-tri, 2023-02-18:
This is working fine for me using Drake v1.13.0 on Ubuntu 20.04:

```console
jwnimmer@cons:~$ wget -qO- https://drake-apt.csail.mit.edu/drake.asc | gpg --dearmor - | sudo tee /etc/apt/trusted.gpg.d/drake.gpg >/dev/null
jwnimmer@cons:~$ echo "deb [arch=amd64] https://drake-apt.csail.mit.edu/$(lsb_release -cs) $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/drake.list >/dev/null
jwnimmer@cons:~$ sudo apt update
...
jwnimmer@cons:~$ sudo apt-get install --no-install-recommends drake-dev
...
Setting up drake-dev (1.13.0-1) ...
...
jwnimmer@cons:~$ dpkg -l drake-dev
...
ii  drake-dev      1.13.0-1     amd64        Model-based design/verification for robotics

jwnimmer@cons:~$ export PYTHONPATH="/opt/drake/lib/python$(python3 -c 'import sys; print("{0}.{1}".format(*sys.version_info))')/site-packages${PYTHONPATH:+:${PYTHONPATH}}"
jwnimmer@cons:~$ echo $PYTHONPATH
/opt/drake/lib/python3.8/site-packages

jwnimmer@cons:~$ python3 -c 'from pydrake.planning import SceneGraphCollisionChecker; help(SceneGraphCollisionChecker)'
Help on class SceneGraphCollisionChecker in module pydrake.planning:

class SceneGraphCollisionChecker(CollisionChecker)
 |  An implementation of CollisionChecker that uses SceneGraph to provide
 |  collision checks. Its behavior and limitations are exactly those of
 |  SceneGraph, e.g., in terms of what kinds of geometries can be
 |  collided.
 |  
 |  Method resolution order:
...
```

closed 2023-02-18 by jwnimmer-tri

ZZHPKU, 2023-02-21:
This really works for me! Thank you very much!