RobotLocomotion/drake #18808: Make Compilation Error：undefined reference to `amd_order'

widthstudio, 2023-02-17:
### What happened?

When I compiled drake's C++ project with make, the prompt "undefined reference to 'amd_order'" was displayed in the terminal. This did not occur when running the python project.

### Version

20220420083150 4e1146e7251aaff1e263b3b10a41f915827d9acb

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

binary tar.gz download

### Relevant log output

```shell
/usr/bin/ld: warning: libamd.so.2, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /opt/drake/lib/libdrake.so: undefined reference to `amd_order'
collect2: error: ld returned 1 exit status
make[2]: *** [src/xxx/CMakeFiles/xxx.dir/build.make:241：src/xxx/xxx] error 1
make[1]: *** [CMakeFiles/Makefile2:1194：src/xxx/CMakeFiles/xxx.dir/all]
```

jwnimmer-tri, 2023-02-17:
Could you explain more about exactly what you are doing, for example the steps to reproduce the problem?

You say "compiling Drake's C++ project" but the error message text `/opt/drake/lib/...` indicates you're using a pre-compiled version of Drake (probably v1.2.0), not building it form source.  What project are you running `make` on?  Is it in Drake, or a project that uses Drake as a dependency?

In that version of Drake (from last year), the `amd_order` symbol should be provided by `/usr/lib/x86_64-linux-gnu/libamd.so.2` which comes from `sudo apt install libamd2`, which should have happened when you ran `share/drake/setup/install_prereqs` from a `drake.tar.gz` release, or automatically if you're using the `apt` release of Drake.

jwnimmer-tri, 2023-02-17:
> Do I need to install Drake from source to solve this problem?

The binary install should be fine.  We just need to see what's going wrong with it.

When installing from tar.gz, you need to run `sudo /pathto/drake/share/drake/setup/install_prereqs` to install what it needs from Ubuntu.  Given your sample paths shown above, I think that would be:

```
sudo /opt/drake/share/drake/setup/install_prereqs
```

That should install libamd2 as well as several other libraries that are required.

widthstudio, 2023-02-17:
> The binary install should be fine. We just need to see what's going wrong with it.
>
> When installing from tar.gz, you need to run `sudo /pathto/drake/share/drake/setup/install_prereqs` to install [...]

When installing, I ran `sudo /drake/share/setup/install _prereqs`under Drake's decompression path.

jwnimmer-tri, 2023-02-17:
Can you post a copy of the exact setup command line that you ran, along with its full output?  (If it's too long, it's fine to put it in an attachment or a https://gist.github.com/.)

According to Drake's [source code](https://github.com/RobotLocomotion/drake/blob/v1.2.0/setup/ubuntu/binary_distribution/packages-focal.txt), it should have installed `libamd2`.

Also can you post the output from `dpkg -l libamd2` please.

widthstudio, 2023-02-18:
> Can you post a copy of the exact setup command line that you ran, along with its full output? (If it's too long, it's fine to put it in an attachment or a https://gist.github.com/.)
>
> According to [...]

[ubuntu.zip](https://github.com/RobotLocomotion/drake/files/10772743/ubuntu.zip)
This is my installation file. When I run 'dpkg -l libamd2', the terminal prompts that there is no matching package.

jwnimmer-tri, 2023-02-18:
Please run `sudo /opt/drake/share/drake/setup/install_prereqs` and see if that fixes the problem.

If not, provide the full console output of that command when you tried to run it.

widthstudio, 2023-02-18:
> 

Thanks for your reply, I have solved the problem now. Thanks!

closed 2023-02-18 by jwnimmer-tri