RobotLocomotion/drake #18816: [multibody] Possible bug in free body dynamics (non-constant axis of rotation)?

EricCousineau-TRI, 2023-02-18:
### What happened?

For writing and debugging spatial controllers, I wanted to ensure that I had the right feedforward math by ensuring that feedforward acceleration should track perfectly with no feedback.

I found that I could track perfectly in C-space for an arm, but I could not track perfectly for either a spatial objective on the arm, nor for a floating body. I found that I could track perfectly if the trajectory had a constant axis of rotation, but not if the axis of rotation was non-constant.

I think this is consistent with the fact that we have the `FreeBody` benchmark, which (I believe) primarily tests a *constant* angular velocity, meaning a constant axis of rotation.

My best attempt to reduce the problem is in #18815. It has reproduction steps shown there, and more information about what it itself does.

At a high level:

- I use Drake symbolics to try two different coordinate representations of SO(3), Euler RPY and Quaternion (as Drake does via `QuaternionFloatingMobilizer`).
- I compute both the generalized position, velocity, and acceleration of the coordinates. I then integrate the generalized accelerations directly (using default variable-step integration via `Simulator`), and ensure I can recover the velocities and positions.
- I then do the mapping stuff that `QuaternionFloatingMobilizer`, but using symbolics just to (hopefully) make as few mistakes as possible.

Assigning @amcastro-tri for disposition
I'm tempted to label this as `priority: high`; however, there is a chance I made a mistake in my SO(3) trajectory stuff.

\cc @mitiguy @sherm1 @RussTedrake given prior convos and stuff

### Version

239f0a4016bfcdcdefa8dd5ef1759a2a4fabe391

amcastro-tri, 2023-02-20:
Do you mean the bench mark in `drake/multibody/benchmarks/free_body/free_body.h`?
If so, that benchmark is for triaxially symmetric bodies, for instance a cylinder or a rod. Since the rotational inertia has two distinct eigenvalues, the angular velocity vector will undergo a precession (motion of its axis of rotation) even though the angular momentum of the free body is conserved (there are no torques applied).

I am sure @mitiguy could confirm or deny this.

EricCousineau-TRI, 2023-02-20:
Yeah, per this and f2f convo, you could be right that my guess on "non-constant axis of rotation" could be wrong.

However, I still think there's something funky going on with my cute 2nd order integration benchmark stuff. Specifically, these parts from 9be12502 in my draft PR #18815:
https://github.com/RobotLocomotion/drake/blob/9be1250294a1a8ed683f20b0f4595c0ad35bb57f/examples/so3_integ_bug/test/spatial_trajectories_test.py#L123-L132
https://github.com/RobotLocomotion/drake/blob/9be1250294a1a8ed683f20b0f4595c0ad35bb57f/examples/so3_integ_bug/test/spatial_trajectories_test.py#L344-L373

EricCousineau-TRI, 2023-02-21:
K, it was as Ale predicted - I was using body-frame angular velocity, not world-frame angular velocity :facepalm: 
52797c2a19

Fixing that makes everything match up as expected!

And yeah, I should've looked at an animation of Paul's analytical solution w/ precessing too - thanks for y'all's patience!

closed 2023-02-21 by EricCousineau-TRI

amcastro-tri, 2023-02-21:
Great job @EricCousineau-TRI! 
I am so happy you figured this one out  :muscle: :muscle: :muscle:  (and there is no serious bug in Drake!)

mitiguy, 2023-02-21:
@EricCousineau-TRI   Quick question -- does that mean you are still working on the feed-forward controller for a free-floating body, or have you already landed one?

EricCousineau-TRI, 2023-02-21:
Working to confirm that the feedforward stuff works on less trivial systems, but I'm fairly confident it should work!
(my "complex" setup would land in Anzu, but I think it'd be easy to test out w/ custom Python code or what not in pure Drake)

EricCousineau-TRI, 2023-02-21:
Free-floating body feedforward acceleration works!
This is checking w/ continuous-time plant, though the rotation inertia is fully symmetric (spherical).

Articulated angular acceleration feedforward works up to a point (near-perfect tracking), but the system seems to get too "stiff" after 1 sec of simulation time. Most likely, just a bug on my part.