RobotLocomotion/drake #18818: [pydrake] Unintended copy in bindings for `DiscreteValues::get_mutable_value()`

RussTedrake, 2023-02-18:
Methods that return an `Eigen::VectorBlock` reference need to have bindings that return an `Eigen::Ref` to avoid a copy (which means that modifications in python do not effect the original data as expected).
 
https://github.com/RobotLocomotion/drake/pull/18809 discovered and resolves this behavior for the particular instance of `DiscreteValues::get_mutable_value()`  (and `get_value`).  But we should seek out any other possible offenders.

In starting to do this, I noticed some inconsistencies in the bindings around this.
- `BasicVector::get_mutable_value()` correctly returns an `Eigen::Ref`.  `BasicVector::get_value()` returns the same.
- `MultibodyPlant::GetMutablePositions()` and friends also correctly returns an `Eigen::Ref`.  But `MultibodyPlant::GetPositions()` returns a `VectorX`, so makes a copy.

Which behavior do we consider to be correct?  (I think we should make them consistent)

RussTedrake, 2023-02-18:
I've looked at all of the instances of `VectorBlock<Vector` in the code base, and the rest all look ok to me.  I would be happy to close this issue once we decide about the inconsistency in `BasicVector` (and now `DiscreteValues`) vs `MultibodyPlant` noted above.

RussTedrake, 2023-02-21:
Sorry.  The point I'm making is that `MBP::GetPositions()` makes a copy, whereas `BasicVector::get_value()` returns a reference (no copy; so it's mutable in pythong).  I view these as being inconsistent.  Both have additional mutable accessor methods.

jwnimmer-tri, 2023-02-21:
It doesn't make a copy, though?  It uses a vector block:

```c++
Eigen::VectorBlock<const VectorX<T>> GetPositions(
    const systems::Context<T>& context) const;
```

jwnimmer-tri, 2023-02-21:
Or do you mean that the _python bindings_ make a copy of that return value?

RussTedrake, 2023-02-21:
right.  the python bindings.  sorry, i thought that was clear.

RussTedrake, 2023-02-21:
If you say that `GetPositions` is correct, then I think the binding for `BasicVector::get_value()` is wrong?
https://github.com/RobotLocomotion/drake/blob/79c5b35b0a265ab2cf2f4f774f72c65b10e00a11/bindings/pydrake/systems/framework_py_values.cc#L97-L114