RobotLocomotion/drake #18830: IRIS using a CollisionChecker

jwnimmer-tri, 2023-02-20:
We currently offer these two functions:

```c++
ConvexSets MakeIrisObstacles(
    const QueryObject<double>& query_object,
    std::optional<FrameId> reference_frame = std::nullopt);

HPolyhedron IrisInConfigurationSpace(
    const multibody::MultibodyPlant<double>& plant,
    const systems::Context<double>& context,
    const IrisOptions& options = IrisOptions());
```

In both, the collision geometry comes from a query object, either directly (in the first) or the `plant` input port (in the second).

The `drake::planning::CollisionChecker` offers a more natural interface for collision queries used for motion planning, and most particularly the ability to configure collision padding on a per-body-pair basis.

It would make IRIS a lot more useful to more people if we could run it using padding, and therefore if it accepted a `CollisionChecker` as an alternative source of geometry and kinematics.

\CC @calderpg-tri @RussTedrake FYI

hongkai-dai, 2023-02-28:
To make sure we are on the same page, I think you are proposing these API

```cc
ConvexSets MakeIrisObstacles(const CollisionChecker& collision_checker, std::optional<FrameId> reference_frame = std::nullopt);

HPolyhedron IrisInConfigurationSpace(const CollisionChecker& collision_checker,  const IrisOptions& options = IrisOptions());
```

In the function `IrisInConfigurationSpace` with a CollisionChecker, I suppose we do not need to pass in a Context<double> object as the input argument? Because CollisionChecker already has a context `CollisionChecker::plant_context()` in it (but the documentation says the function might be tossed)? https://github.com/RobotLocomotion/drake/blob/6f1a88111c7b04217d461e906521a9ac7f6e58ed/planning/collision_checker.h#L239-L245.

jwnimmer-tri, 2023-02-28:
The documentation of the existing `IrisInConfigurationSpace` says that the `initialIRIS seed configuration` is passed via the Context.

In the new function, I think asking the user to call `CollisionChecker::UpdatePositions` to pass the seed would be awkward at best. Perhaps the new function can just take an Eigen `q` vector as argument for the seed.

sadraddini, 2023-03-01:
```

> ConvexSets MakeIrisObstacles(const CollisionChecker& collision_checker, std::optional<FrameId> reference_frame = std::nullopt);
> 
> HPolyhedron IrisInConfigurationSpace(const CollisionChecker& collision_checker,  const IrisOptions& options = IrisOptions());

```

Looks good. I think passing `const Eigen::VectorXd q_seed` is more natural than passing the context for the first.

RussTedrake, 2023-03-01:
To be clear, the normal preference for passing a context instead of just a state is to handle additional features like parameters. If this is not needed here (e.g. if the geometries/kinematics are all specified already at the time of the call, so the effect of parameters is irrelevant), then q_seed should be sufficient.  Otherwise we are losing something.

I admit that supporting changes in kinematics through the parameters might not be a big loss. Just want to make sure you understand Chesterton's fence.

jwnimmer-tri, 2023-03-01:
It's a really good point.  If users need non-default kinematic parameter for their motion planning, I think we need to teach CollisionChecker to handle that directly (by letting users set non-default Parameters on the checker itself).

calderpg-tri, 2023-03-01:
Are there parameters that couldn't be set through the existing [`CollisionChecker::PerformOperationAgainstAllModelContexts`](https://github.com/RobotLocomotion/drake/blob/master/planning/collision_checker.h#L278) mechansim?

jwnimmer-tri, 2023-03-01:
Great point.  That's precisely the way to do it, and will cover all cases.  All parameters are sufficiently provided by the `CollisionChecker`.  That (plus a `q` vector argument for convenience) should be good enough:

```c++
ConvexSets MakeIrisObstacles(
    const CollisionChecker& collision_checker,
    std::optional<FrameId> reference_frame = std::nullopt);

HPolyhedron IrisInConfigurationSpace(
    const CollisionChecker& collision_checker,
    const Eigen::Ref<const Eigen::VectorXd>& q_seed,
    const IrisOptions& options = IrisOptions());
```

hongkai-dai, 2023-03-05:
BTW, I would think both `MakeIrisObstacles` and `IrisInCOnfigurationSpace` would need `q_seed`? Currently `MakeIrisObstacles` documentation says https://github.com/RobotLocomotion/drake/blob/42d86973c41725ad4ef1848547374fa21263009f/geometry/optimization/iris.h#L133-L136, I think we also need a configuration to fix the geometry poses in `MakeIrisObstacles`.

hongkai-dai, 2023-03-06:
Also I would like to make sure that we really need this function `MakeIrisObstacles` that takes `CollisionChecker` as an input argument. As far as I can see the implementation is just a two-liner
```cc
ConvexSets MakeIrisObstacles(const CollisionChecker& collision_checker, const Eigen::Ref<const Eigen::VectorXd>& q_seed, std::optional<FrameId> reference_frame) {
  collision_checker.UpdatePosition(q_seed);
  return MakeIrisObstacles(collision_checker.model_context().GetQueryObject(), reference_frame);
}
```
if the implementation is really this short, why the user cannot just type it themselves?

jwnimmer-tri, 2023-03-06:
> if the implementation is really this short, why the user cannot just type it themselves?

The query object does not account for the collision checker's configured padding matrix.

calderpg-tri, 2023-03-06:
More importantly, IRIS using a `CollisionChecker` provides for IRIS using a collision checker that is not `SceneGraphCollisionChecker` (e.g. Anzu's `VoxelizedEnvironmentCollisionChecker`), where the robot and/or environment geometry is not necessarily part of the collision checker's SceneGraph.

hongkai-dai, 2023-03-08:
Sorry for being silent on this thread.

Sorry I am very new to CollisionChecker class. My understanding is that CollisionChecker contains two types of geometries, the "model geometries" and "checker geometries". `collision_checker.model_context().GetQueryObject()` contains both model and check geometries.  What it doesn't contain is the padding for each pair of geometries.

Padding is added for each pair of geometries. On the other hand, in `MakeIrisObstacles`, we want to convert each geometry (for example `G1`) to a convex set, which would be the same convex set for any pair of geometries (G1, G2), (G1, G3), ... I think I am confused here on how to handle the padding.

calderpg-tri, 2023-03-08:
Only in `SceneGraphCollisionChecker` does the context contain all geometries. For any other implementation of `CollisionChecker`, the robot and/or environment geometries being considered could be arbitrarily different from those in the context.

While it is possible to extract the complete robot and environment geometry (in different ways) from every extant collision checker, that's not the intended use. What we are really talking about here is IRIS using `CollisionChecker` _queries_ (i.e. binary and distances), rather than its internals.

jwnimmer-tri, 2023-03-08:
We might need an offline sync here.  That doesn't match with my understanding.

hongkai-dai, 2023-03-08:
> While it is possible to extract the complete robot and environment geometry (in different ways) from every extant collision checker, that's not the intended use. What we are really talking about here is IRIS using CollisionChecker queries (i.e. binary and distances), rather than its internals.

I can see how that works out for IrisInConfigurationSpace (it would require writing a bunch of functions, such at `SamePointConstraint` that takes CollisionChecker instead of a plant and a context). But I don't see how that works for MakeIrisObstacles, which really only need the collision geometry of each individual geometry.

I have sent a hangout invitation, we can sync up then.

sadraddini, 2023-03-09:
> I can see how that works out for IrisInConfigurationSpace (it would require writing a bunch of functions, such at SamePointConstraint that takes CollisionChecker instead of a plant and a context). [...]

I agree with this. I was confused about what `MakeIrisObstacles` is - I thought it is about C-space. When it comes to configuration in workspace paddings of collision checkers are not relevant.

jwnimmer-tri, 2023-03-09:
@sadraddini Ok, thanks!  I'll re-scope this issue to just the `IrisInConfigurationSpace` function.

---

@hongkai-dai @calderpg-tri for our chat later today, refer to this section of code from `IrisInConfigurationSpace`:

https://github.com/RobotLocomotion/drake/blob/0cb9faa6902e2c4558f33b35d2ef1cdbe05001fe/geometry/optimization/iris.cc#L643-L661

We convert every geometry to a convex set.  We can easily switch that from every GeomtryId to every BodyIndex, that's not difficult.  But when per-pair padding, do we need per-pair convex sets now (possibly memoized if padding if often the same)?  And how to we extract the voxel grid from the voxel checker and make it into a convex set?

Or is that the wrong idea, and we just need to implement it from scratch as a nonlinear program that only ever does signed distance queries, with no convex sets?  Wouldn't that defeat the purpose?

We can chat in a few hours.

calderpg-tri, 2023-03-09:
> Or is that the wrong idea, and we just need to implement it from scratch as a nonlinear program that only ever does signed distance queries, with no convex sets? Wouldn't that defeat the purpose?

This is my understanding. The purpose is to produce the same kinds of C-space collision-free regions as the existing `IrisInConfigurationSpace`, just using `CollisionChecker` queries instead of knowledge of the specific collision geometries. An advantage of this formulation is that no additional work is required to support any additional geometry types supported by other `CollisionChecker` implementations (e.g. voxel grids, octomap, etc).

jwnimmer-tri, 2023-03-09:
FYI @hongkai-dai on the distance query functions:
* [`MaxNumDistances`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1_collision_checker.html#a4f0eb0fde313a6653e5692173c759145) to find out how many distance measurements we might need to optimize against.
* [`CalcRobotClearance`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1_collision_checker.html#a678d1e0c7b171295455bea0b60ab85d3) which returns a table of [`RobotClearance`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1_robot_clearance.html) for the actual distances, using a given `q`.

hongkai-dai, 2023-03-10:
Current CollisionChecker `CalcRobotClearance` function returns a matrix for the distance in every pair of geometries. On the other hand, in IRIS we would like to solve many small mathematical programs, each mathematical program only considers one pair of collision geometry. Is it possible to have `CollisionChecker` to compute the clearance for a specific pair of geometries?

jwnimmer-tri, 2023-03-13:
We can't go down to the level of _geometry_ (i.e., shapes) with a CollsionChecker.

We could probably offer a query that returns a subset of the clearance, filtered either for:
(1) measurements that mention a particular BodyIndex (e.g., the end-effector collising with anything else, either robot or environment)
(2) measurements that mention _two_ specific BodyIndex values (e.g., link6+link7, or link6+world, etc.).

It sounds like what you'd propose doing is:
(a) Get the full clearance table.  Sort by distance (just like we do today -- "As a surrogate for the true objective, the pairs are sorted by the distance between each collision pair from the sample point configuration...").
(b) Loop over the body pairs in the clearance table, calculating clearance for just that one body-pair as part of the nonlinear optimization program.

In that case, you'd want API function (2) as part of step (b) -- 

```c++
RobotClearance CalcRobotClearance(
    const Eigen::VectorXd& q,
    std::pair<BodyIndex, BodyIndex> bodies,
    double influence_distance) const;
```

hongkai-dai, 2023-03-13:
The more granularity we have in computing the signed distance, the better it is for the IRIS nonlinear program. So I would say (2) (signed distance between two bodies) is better than (1) (signed distance between a body and everything else).

What I have in mind now is that I will write a constraint
```
min(CalcRobotClearance(...)) < 0
```
The `min` operation will be problematic for nonlinear optimization, as it does not have a smooth gradient when `a=b` in `min(a, b)`. I could use soft_min instead of `min`, but that only alleviates the gradient problem, doesn't completely resolves the issue. The less entries I have in the `min` function, the nonlinear optimization probably works better (hence the best situation is that CalcRobotClearance returns one distance for a single pair of geometries, but as you said that is not doable).

jwnimmer-tri, 2023-03-13:
Even if we had a query for low-level pairs of shapes, I think the gradient of closest distance would not always be smooth?  The shortest distance segment between two shapes can still jump between different faces / edges / points as we reposture things.  It's just intrinsic to the problem of using anything except extremely primitive shapes like ellipsoids for proximity.

hongkai-dai, 2023-03-13:
Right, when one geometry crosses the median axis of the penetrated geometry, then the distance is non-smooth. My understanding is that sometimes we do use spheres as collision avoidance hence we do not need to worry about the median axis problem. Nevertheless, I think if we can reduce the number of non-smoothness then the code could run better. I think if CollisionChecker can return a table of clearance between to bodies that should be a good starting point.

jwnimmer-tri, 2023-03-15:
Yup.

If we wanted to test the premise (but with slower runtime perf), we could do that today with the current Clearance API.  We'd just filter by body-index in the IRIS code, instead of passing the body index as an argument to a new API.  It would run a lot slower, but I think it would still pass all the unit tests and we could see if SNOPT likes the gradients?

hongkai-dai, 2023-06-07:
Sorry I was silent on this issue for a while.

Currently we do not have the API to return the clearance between a specific pair of bodies. So I will use `CalcRobotClearance` which computes the distance between every pair of bodies, and then select out the specified body pair in the new IRIS implementation. If the SNOPT is happy with this body-wise distance, then we can add the API to CollisionChecker.

hongkai-dai, 2023-08-30:
I just had a meeting with Tom and Akio from Woven, they are interested in using IRIS with a TSDF. I think once we have the IRIS-NP that uses CollisionChecker we should be able to support their use case.

I can push on this.

hongkai-dai, 2024-08-22:
I missed the action item for this issue in today's GCS standup meeting. I suppose this issue is still relevant, even if we work on #21822? Do we want to wait for #21821 (the refactorization) to finish and then work on using CollisionChecker?

RussTedrake, 2024-08-22:
The plan from today, as I understand it, is to commit #21822 with CollisionChecker in the api, but without supporting many of the CollisionChecker features that we most wanted. So this issue is still relevant, but I think the goal should now be to support IRIS-NP2 instead of IRIS-NP.

cohnt, 2024-08-23:
Just some additional detail -- IRIS-ZO should support any CollisionChecker pretty much out of the box (@wernerpe correct me if I'm wrong).

IRIS-NP2 will likely only be able to support SceneGraphCollisionChecker, due to the need to construct the counterexample programs based on individual robot geometries (voxels would be very inefficient if not outright intractable, for example). Of course, one could imagine tracing the gradient of the C-space signed distance function to abstract away from obstacle pairs at some point in the future.

Also, there are a couple features in CollisionChecker that may be challenging to include in IRIS-NP2. Negative padding is the big one that comes to mind -- I don't know how to do that in the counterexample search program. (But positive padding should be fine.)

RussTedrake, 2024-08-23:
I understood that the work Hongkai had imagined here would have been to support the min distance queries directly in the collision checker base class (for all implementations). @hongkai-dai -- maybe you can weigh in?

hongkai-dai, 2024-08-24:
Right, I was imagine to use the CollisionChecker class to find the closest in-collision configuration, which was used in the original IRIS-NP algorithm. I tried to work on it a while ago, but the MinimumDistanceUpperBoundConstraint doesn't work as nicely as I expected (somehow it can fail to find the in-collision configuration in a toy example with just two spheres.)

calderpg-tri, 2024-09-10:
> closest in-collision configuration ... but the MinimumDistanceUpperBoundConstraint doesn't work as nicely as I expected ...

Depending on the behavior you want, e.g. if _a_ nearby in-collision config, rather than _closest_ in some sense, you might be able to use the gradient from [ComputeCollisionAvoidanceDisplacement](https://github.com/RobotLocomotion/drake/blob/master/planning/collision_avoidance.h#L53) to iteratively adjust from a free to colliding config. In my experience, manually iterating with `ComputeCollisionAvoidanceDisplacement` to go from colliding->free config is significantly faster (although somewhat less likely to succeed) than posing the same projection as an optimization problem. I suspect that the same may hold for free->colliding.