RobotLocomotion/drake #18839: YCB Collision Geometry Sphere Placement

nepfaff, 2023-02-21:
### What happened?

I looked at the [YCB mustard bottle collision geometry](https://github.com/RobotLocomotion/drake/blob/master/manipulation/models/ycb/sdf/006_mustard_bottle.sdf) and noticed that the spheres are offset from the main box primitive. Wouldn't the desired placement be at the box edges?

A collision geometry visualization with the sphere radii inflated from 1e-7 to 1e-3 for visibility:
[image]

jwnimmer-tri, 2023-02-21:
+@EricCousineau-TRI is there any history here we should know about?

EricCousineau-TRI, 2023-02-23:
Nothing that I'm aware of, unfortunately.

jwnimmer-tri, 2024-03-20:
Re-filed in new home of YCB objects: https://github.com/RobotLocomotion/models/issues/39

closed 2024-03-20 by jwnimmer-tri