RobotLocomotion/drake #18842: Use --config=packaging in drake-ci

jwnimmer-tri, 2023-02-21:
Towards #18828 and continuing the work of #18840, I'd like to rename the https://drake-jenkins.csail.mit.edu/view/Packaging/ builds.

(1) I'd like us to change the drake-ci driver to notice "Packaging" in the job name and pass `--config=packaging` to Bazel in that case.

(2) I'd like to delete `-snopt-mosek` from all of those job names.

(3) Update the release playbook and download_release_candidate to reflect the new names.

TTBOMK we need to do either (1) then (2) in sequence, or else (1+2) concurrently.  We can't do (2) first, or else we'd lose snopt and mosek support in those builds.

jwnimmer-tri, 2023-02-22:
FYI See #18849 as the intended consumed of this change.  Feel free to use it for testing in case that helps.

BetsyMcPhail, 2023-02-22:
@jwnimmer-tri should the [SNOPT](https://drake-jenkins.csail.mit.edu/view/SNOPT/)and [MOSEK](https://drake-jenkins.csail.mit.edu/view/MOSEK/) still list the packaging jobs?

jwnimmer-tri, 2023-02-22:
That's a good question.

To me, I think it's fine to drop them from those tabs.  The packaging jobs aren't actually running any tests, so I don't know that they are adding much to those tabs.

If y'all agree, let's say drop them.

BetsyMcPhail, 2023-02-22:
That's my preference as well, I will drop them. Or, more precisely, not change the current regex to explicitly add them back in after they've been renamed.

BetsyMcPhail, 2023-02-23:
As required by Jenkins, I launched the new Continuous job by hand: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-unprovisioned-gcc-bazel-continuous-packaging/. From now on, this should launch automatically as expected.

jwnimmer-tri, 2023-02-23:
I'm going to reopen to add a related feature request.  For the wheel builds: 

https://drake-jenkins.csail.mit.edu/view/Wheel/

... let's please do likewise, to remove "snopt-mosek" from the job names.

jwnimmer-tri, 2023-03-02:
Yes, exactly.  The wheel build already had (has) local authority over its build flags.  The drake-ci code feeds in version numbers and directory names, but not flags.

PR #18851 (2023-02-23): [doc] Update packaging job names in the Release Playbook
Resolves #18842, blocked by https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/66

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18851)
closed 2023-03-07 by jwnimmer-tri