RobotLocomotion/drake #18844: Texturing for Drake supported RenderEngine and visualizers

zachfang, 2023-02-21:
The goal of this epic issue is to **summarize** and **prioritize** existing issues (usability, consistency, missing feature) regarding `primitive` and `mesh` texturing in either Drake's supported renderers, e.g., `RenderEngine{Vtk, Gl, GltfClient}`, or visualizers, e.g., `MeshcatVisualizer` and `Meldis`.

The ultimate goal is to **(1) support common texture-loading scenarios** and to **(2) have consistent texturing behavior** across renderers and visualizers in Drake. Along the way to achieve the goal, we should improve the documentation and code (warning or throw) to inform users of the current status, i.e., what is supported and what is not.

Backlog:
- [ ] RenderEngine(s) should support loading mtl file (#11949).

zachfang, 2023-02-21:
To break down the problem, these are the existing or common ways to specify a texture (IMO).
1. A single phong-like diffuse color.
2. `OBJ-to-PNG` association in the same folder.
3. Specify a texture through `<drake:diffuse_map>` tag via parsers.
4. Specify texture(s) through an MTL file.

- Drake's visualizers (only tested with Meldis for now) support (1) and (4).
  - For (3), `diffuse_map` is only used in `perception` roles only.
  - Note that primitives are not textured in the visualizers.

- Drake's renderers (only tested with RenderEngineVtk for now) support (1), (2), and (3)*.
  - `<drake:diffuse_map>` is [only parsed via SDFormats but not URDFs](https://github.com/RobotLocomotion/drake/blob/1397c24defca001f169dfddaa7700b8f2deb6abb/multibody/parsing/parsing_doxygen.h#L414-L418).

zachfang, 2023-02-21:
For demonstration purposes, five meshes (base, shoulder, upperarm, forearm, and wrist1 of an UR3e) are loaded through an SDFormat and a URDF. The visualization results from a visualizer (Meldis) and a renderer (RenderEngineVtk) are compared side-by-side.

Five meshes all use different texture specifications.
- Base: Textured with the referenced MTL file with the same name in the OBJ.
- Shoulder: Textured with the referenced MTL file with a different name in the OBJ.
- Upperarm: No MTL file. Textured with a simple diffuse color, i.e., RED.
- Forearm: No MTL file. Textured via `<drake:diffuse_map>` tag in the SDFormat or URDF file.
- Wrist1: No MTL file. Textured via a PNG file with the same name.

The result from the URDF:
[image]

The result from the SDFormat:
[image]

zachfang, 2023-03-23:
To be explicit, I am now working on (1) from the above backlog. Will update my concrete approach in this comment once I experiment with a few options from #11949.

I am also planning to work on (2). My thought is to:
1. Add some documentation [here](https://github.com/RobotLocomotion/drake/blob/1397c24defca001f169dfddaa7700b8f2deb6abb/multibody/parsing/parsing_doxygen.h#L414-L418) (or some relevant places).
2. Warn users if a `diffuse_map` tag is found when parsing a URDF and remind them to use the URDF built-in `<material/texture>` tag (as Sean pointed out) instead. Currently, it is ignored silently.

SeanCurtis-TRI, 2023-03-27:
@zachfang based on today's meeting, I'm going to break up (1) to enumerate the render engines separately.

jwnimmer-tri, 2023-03-28:
I'm going to work on the `Meldis` tasks, but they are mostly blocked until `MeshcatVisualizer` is finished first.  Until I can pass a `diffuse_map` property to `MeshcatVisualizer` and have it show up in the display, there's little I can do.  (I could copy the obj into a tmpdir and insert an mtl file to specify the map, but that's silly when the correct property handling is underway.)

SeanCurtis-TRI, 2023-03-28:
I had some random thoughts:

The table above could actually be partitioned into two categories:

1) Geometries (typically meshes) that specify their own materials (including textures) in their on-disk representation.
2) Geometries that have a texture map associated with them in a `GeometryProperties` instance.
    - This second category is taking a *slight* liberty by subsuming the "there's a png that matches the obj" logic. However, the thought that occurred to me is that it *should* subsume that logic. When registering a `Mesh`, `SceneGraph` should look for the png and add a property indicating the presence of a co-located png. Then consumers of the shape and its properties wouldn't have to *implement* the logic of "look for the png", it would simply say, "Ah....SceneGraph has identified a candidate fallback diffuse texture."

This also directly relates to all of the visualizers that currently fail to load a diffuse map from `GeometryProperties`. We solve that transport issue once, we get the solution every time. And, then, it's just a case of worrying about getting disparate OBJ parsers to treat OBJs the same.

SeanCurtis-TRI, 2023-03-28:
Another topic we'll need to address:

Confusing support for OBJs. In a recent case, we had an OBJ exported from blender. The material was PBR-style. However, OBJ doesn't really support it. Upon loading in `ModelVisualizer` some textures played no role at all, some played a confusing role, and the parameters remaining that contributed to the end result were inappropriate.

That means we probably need to do some vetting/warning at each geometry consumer. It's not enough to make sure the MTL file is consumed, we should also look at it and say, "Hmm....you've got stuff in here that will be ignored or misinterpreted."

For example, bump maps:
    - `ModelVisualizer` supports them, but
        - Exported from blender they are frequently *normal* maps and not *bump* maps which lead to a qualitative difference in appearance.
        - The bump map scale will probably be *wildly* inappropriate (this is one we could hope to address with some man-in-the-middle processing).

It's worth noting, that we can't simply rely on what is considered valid for an MTL file, but what is considered by the various parsers we're subject to and how mesh *sources* may interpret valid texture declarations differently. (Blender maps PBR "roughness" to "specular shininess" and "metalness" to "reflection" which will not be equivalent and may require massaging to become a meaningful Phong material.)

SeanCurtis-TRI, 2023-03-29:
A treatise and proposal for the semantics of defining the properties ("phong", "diffuse") and ("phong", "diffuse_map").

__Background__

- URDF: URDF [supports materials](https://wiki.ros.org/urdf/XML/link#line-80). It include an optional "color" (rgba) or "texture".
   - It is *not* documented how those two quantities relate if both are provided.
   - There seems to be no standard provided by URDF.
- SDF: SDFormat files support a [host of material data](http://sdformat.org/spec?ver=1.9&elem=material).
    - SDFormat's material is simultaneously more elaborate and more opaque than URDF.
    - The material supports a *normal_map* but only rgba values for  ambient, diffuse, specular, etc. (the phong material properties)
    - It does have a PBR specification including constant values as well as maps.
- Material definitions
    - For common lighting models, the "albedo" (or diffuse color) of a surface is typically the product of two values: the underlying diffuse color and the sampled color from the optional diffuse map. To reproduce the texture exactly, the diffuse color is usually opaque white (1, 1, 1, 1). However, it is a common graphics trick to change the underlying color to modulate the end color -- tinting the end result. This works well for creating visual variance across a number of instances of a single geometry, material combination.

__Drake history__

It seems convenient to be able to assign a texture a visual geometry in the model specification. URDF provided a very simple mechanism and the `<drake:diffuse_map>` tag was added to SDFormat to provide parity.

One of the primary benefits is associating a texture with a *primitive* shape -- a geometry specification that doesn't encode its own materials. Rather than merely distinguishing the shapes with simple colors, we could also use textures (good for creating simple models on the fly like wooden tables). There was a vague intent to expand the set of parameters to further include a more elaborate material specification but it has never come to pass.

When diffuse color or a diffuse map is specified in the model specification file, it lives in Drake as a geometry property (both illustration and perception) called ("phong", "diffuse") and ("phong", "diffuse_map"), respectively. Similarly, the user can use a public API to introduce the properties. When the geometries are consumed, there is no ability (need) to distinguish which of these two sources gave rise to the presence of the properties.

__The problems__

1. Texture coordinates
    - Texture coordinates on primitives
        - Drake has worked hard to create objects with texture consistent coordinates across its internal render engines. (Even going so far as to create [custom VTK types](https://github.com/RobotLocomotion/drake/blob/master/geometry/render_vtk/internal_render_engine_vtk_base.cc#L57) to provide a consistent set of texture coordinates different from VTK's defaults. (Note: VTK's capsule is missing those UVs which is being tackled by #18547).
        - The visualizers we use make no guarantees as to the presence or nature of texture coordinates for primitives. The appearance of such a primitive in in the visualizer will be a poor representation of what it would look like in a Drake `RenderCamera`.
    - If texture coordinates are missing, a diffuse map (or any other kind) cannot be applied. But the geometry should still be visible.
2. Combining diffuse color and diffuse map
    - Drake has no standard for how these are combined (and probably no consistency in implementation).
3. Combining with geometries that have their own materials.
    - Primitives don't define materials. General meshes can/may.
        - OBJs come with mtl files, gltf and dae files can contain their materials in the same file as geometry. STL files provide nothing.
    - General meshes are not obliged to consist of a single material. The triangles can be partitioned arbitrarily with different materials to each partition.
    - There are a number of choices that can be made when reconciling define ("phong", "diffuse") and ("phong", "diffuse_map") values with the intrinsic materials:
        - Simply replace *all* materials with a single new "Drake default phong" material consisting of the values stored in the geometry properties.
        - Apply the values defined in geometry properties to every existing material. This would preserve *other* material properties (e.g., specularity, metalness, etc.)
        - Conservatively *replace* values where they exist. So, if a material *doesn't* have a diffuse (or albedo) map already, it won't get one.

__The proposal__

1. Resolving UV issues
    - When Drake internals can't provide UVs, declarations that request a texture map should provide a warning.
        - This includes if we have a substandard geometry (e.g., Capsule right now) or if the user provides a mesh that is missing UVs.
        - Such an object should be rendered with a *white* material (or, a user-configurable value to help call out that it is a defective geometry specification).
    - Document Drake texture coordinates on primitives.
        - Make this clear for external renderers so they can choose to be compliant.
        - Communicate that compliance is optional, so a rendered image may change from a Drake render engine to an external render engine based on whether it creates the same kind of texture coordinates.
[... cut ...]
2. Combining diffuse and diffuse map (with requisite documentation)
    - If only diffuse is given, the material is a solid color.
    - If only diffuse map is given, the final diffuse color is the diffuse map (essentially, the underlying diffuse color is white)
    - If both are given, the diffuse color modulates the diffuse map.
3. Specifying a geometry property completely replaces *all* previously-existing materials in a mesh file. There should be no attempt to combine in any way. The end result on an arbitrary mesh would be the same as if were a primitive.

jwnimmer-tri, 2023-03-29:
> Likewise document that the texture application of primitives in visualizers may not match what is observed from a Drake-internal RenderEngine. This is a very dissatisfying answer, particularly for [...]

This is pretty easily fixed on the ModelVisualizer side -- I've always planned & expected that the model visualizer will provide a live image preview of the render engine output (e.g., using #18862).  So while I agree _in the ideal_ that Meshcat should use the same UV convention for primitives as Drake, in practice I am unsure whether or not it will be a problem.  I'd propose waiting for the "camera preview" feature to land in model_visualizer, and then re-assess whether to do more on Meshcat / MeshcatVisualizer itself.

> Specifying a geometry property completely replaces all previously-existing materials in a mesh file. There should be no attempt to combine in any way. The end result on an arbitrary mesh would be the same as if were a primitive.

I'm 100% on board with the "never the twain shall meet" aspect of this proposal.  For a mesh coloring & texturing specification, users should either use the XML file xor the intrinsic mesh file format -- they cannot mix & match (for the same mesh).

The remaining points of debate would be: which one wins, and what conditions are warnings or errors?

I am tentatively on board with "XML always wins".  I think it's probably the best semantics, by a nose.  My worry is how difficult it would be to implement.  It seems to me like it implies rewriting mesh file contents before passing it along to a consumer.  And in general (e.g., once we support loading glTF meshes) that seems like a giant pain in the butt.

How much of a problem do you think it would be to choose "intrinsic texture always wins"?  In that case, we would warn in the Parser that we're ignoring the XML-given texture specification because the mesh already had one in-band.  That seems to me like it would be a lot easier to implement.

SeanCurtis-TRI, 2023-03-29:
> I'd propose waiting for the "camera preview" feature to land in model_visualizer,

Agreed. This is one of the ones I was going to defer in my current branch.  :)

> How much of a problem do you think it would be to choose "intrinsic texture always wins"? 
 
From a render engine perspective, I don't think it's particularly hard. VTK does the heavy lifting of gltf parsing. It's then a case of manipulating the in-memory VTK representation of swapping out materials. The GL engine is more work. Not because we need to replace the mtl definition, but because we don't have it at all. But replacing it should be just as easy (we swap at parse time).

`MeshcatVisualizer` is straightforward. It reads in the OBJ and puts it in the wire. My first hacky pass was manipulating the data on the wire at the texture level. Manipulating the actual OBJ specification is pretty easy. And since it's only done when the geometry is registered, the string manipulation isn't onerous.

What I'm most worried about is that my patch to Meshcat (as you'll see later today) consists of passing a whole GeometryProperty through instead of just "rgba". THat works well for `MeshcatVisualizer`, but I'm not sure what Meldis is going to do. If I recall correctly, it's merely capturing LCM messages and that is an impoverished data set.  :-/

I'll get my PR opened in a bit and we can use that as a starting point to discuss things.

jwnimmer-tri, 2023-03-29:
If we are trying to promise "properties always win over mtl" then also in scope are teaching [RvizVisualizer](https://github.com/RobotLocomotion/drake-ros/blob/main/drake_ros/viz/rviz_visualizer.h) and every other visualizer end-users need to write how to do this.  It's partially that each individually are difficult, and partially that we're going against the grain of how other toolboxes expect things to work, so each one will require special handling.  I am against solutions that require special handling in every consumer of scene graph. If we are deciding that properties trump mtls (and gltfs), then we need to have SceneGraph rewrite the file, not every single point-of-use.

SeanCurtis-TRI, 2023-03-29:
That is certainly the most robust solution.

So, that suggests the following:
  1. SceneGraph creates a cache directory (e.g., `FindOrCreateCache`).
  2. When "visual" meshes come through, it creates alternates as necessary (placed in the cache and primitives reference *those* files.)
  3. In this case of error messages, they'd naively report about the alternative obj (which implies something along the lines of naming the alternate in terms of the original). Unless we want to get aggressive about hiding mapping. I like the renaming, because it will contribute to clear communication that we're using a mutated version of their obj.

Am I missing anything?

jwnimmer-tri, 2023-03-29:
What you're missing is that I still think that it's not worth the effort.  Especially once we start supporting glTF, or other more complicated mesh import formats.  I think "intrinsic file always wins" is so extraordinarily easy to implement, and so natural to explain to users, that we should choose that as a solution.

> I am tentatively on board with "XML always wins".

I'll retract this statement.  I think it's too complicated.

MTL should win.  We should (ideally) warn when the XML has materials that will be ignored due to MTL.  But since in all of Drake and Anzu there are a total of **zero** files that tried to do that, I don't think it's urgent.

jwnimmer-tri, 2023-03-29:
> I'm not sure what Meldis is going to do.

=> #19077

SeanCurtis-TRI, 2023-03-29:
Ah....

I see your point. The warning is "the mesh has defined its own materials", so the properties will be ignored. That's never relevant for primitives (as there can be no conflict). So, anything existing in geometry_properties should only be applied if a **mesh** doesn't have materials.

However, that takes us right back. If a mesh is created without materials, the declared geometry properties should apply. The only reliable way to get all visualizers to respect that would be to add the materials ourselves (come back to cached alternate versions).

Alternatively, at parse time, we say that any supported mesh format that supports material definitions *require* materials defined in that manner and attempts to set them in the `GeometryProperties` sense is considered an error (maybe warning). Although, doing it at parse time means anyone exercising the API won't get that warning. But putting the warning in the API won't have the parsing context, so it wouldn't respect the parsing diagnostic policy without some extra work.

jwnimmer-tri, 2023-03-29:
> If a mesh is created without materials, the declared geometry properties should apply.

Yes. 

I could maybe buy that something like glTF would unconditionally ignore XML materials (perhaps under the umbrella of "an incomplete implementation") if it's difficult to realize.

> The only reliable way to get all visualizers to respect that would be to add the materials ourselves (come back to cached alternate versions).

No.  It's fine to assume that visualizers should obey properties, and that if they don't then its an implementation deficiency in the visualizer implementation itself.  The only thing I want to avoid individual visualizers knowing about is _our arcane heuristics_ for how to reconcile conflicting advice.  We can accomplish that by simply not converting XML properties into geometry properties in scene graph, when they are in conflict.

SeanCurtis-TRI, 2023-03-29:
BTW I just got it working in meshcat.

There are two kinds of boxes (one with mtl file and one without). The bottom, bright green box is the obj in its native format. The boxes directly above it are the same box with overriding diffuse map and overriding diffuse color. The second column is a box with no mtl in the obj.

[image]

SeanCurtis-TRI, 2023-03-29:
Also, this suggests we need to revisit the motivating table above. I believe that was quite mesh centric. Based on the evolving conversation, we have one axis consisting of the render technologies (render engines and visualizers) and the other axis would be predicated on the (geometry-type, material origin) combination. While that combination generally implies a large matrix, in practice, I think it devolves two two or three categories:

1. A mesh of supported type (OBJ today, tomorrow others) that has material definitions should come through in every context (as well as we can).
    - When this happens, it can still lead to surprising visuals where it isn't obvious what is specified has come through. That is attributable to mesh provenance and compatibility. Being able to predict this is great. Adding to the troubleshooting guide where we can point people would be good.
2. Primitives with ("phong", "diffuse*") properties (color or map) should have *that* material come through in every context (as well as we can).
   - Currently true for diffuse *color* but not *map*.
   - This can be challenging just because `DrakeVisualizer` messages don't support textures.
   - Meshcat can handle this far more easily (it can be passed the properties *and* is responsible for broadcasting all of the data over the wire).
3. As yet undetermined, specific meshes that *haven't* specified materials can/can't have ("phong", "diffuse"*) properties assigned.

SeanCurtis-TRI, 2023-03-29:
I guess my point is, the issue has gotten to a point that I don't think a reader can infer what is broken and what needs to be fixed (I include myself in that category).  The table need not be reworked, but it no longer serves the purpose. There are boxes that currently say "Correctly rendered" that, as near as I can tell, are no longer correct. And vice versa.

So, when I consider `MeshVisualizer`, I'd enumerate the following defects:

1. Primitives with ("phong", "diffuse_map") aren't getting their maps.
2. There are textures supported by Meshcat and OBJ that aren't being transported (meaning that the visualization will be impoverished).
3. Maybe an obj without a material library *should* respect both ("phong", "diffuse") and ("phong", "diffuse_map").

And that's it. Is there a more succinct description of the defect and the victory condition?

jwnimmer-tri, 2023-03-29:
Yes, but also:

4. The "magic png filename" heuristic is being ignored by MeshcatVisualizer.

For that, we can decide either to implement it everywhere, or deprecate it in the render engines.  I'm leaning towards deprecation, subject to everyone's feedback.

SeanCurtis-TRI, 2023-03-29:
I'm in favor of deprecating it.

If we were to not deprecate it, I assume that the image would only be used if:
  - The geometry is a mesh.
  - The geometry has no materials defined of its own.
  - There is no ("phong", "diffuse*") geometry property defined (i.e., just defining a color would be enough to disable the heuristic).

Therefore, I'm excluding this heuristic from my statement of work in anticipation of the new and glorious future.

SeanCurtis-TRI, 2023-03-30:
I think that's only a partial solution. While it removes the arcane knowledge of whether they *should* override it or not (isolating the logic to `SceneGraph`), it still behooves each and every implementation to have to implement special logic in processing objects. For primitives, that's fine, but for meshes that's still where the significant labor resides. For Meshcat, you can see the [tip of the iceberg in my branch](https://github.com/SeanCurtis-TRI/drake/blob/PR_meshcat_texture_warnings/geometry/meshcat.cc#L319-L323). And this is only possible because I'm willing to interfere with the definition of the OBJ. Other consumers may be less willing to do so.

So, I'm in favor of doubling down on the rule: if the geometry format you're using supports materials, use its mechanism. Drake's materials are for those that lack that expressive power. If `SceneGraph` simply presents the data required to operate in an obvious manner, the whole universe is a better place.

SeanCurtis-TRI, 2023-05-10:
Note to self: with the conclusion of #19383, `RenderEngineVtk` has been brought up to the new, short-term standard regarding material definitions. Update the table and statement of work above to reflect this.

zachfang, 2023-06-22:
@SeanCurtis-TRI I plan to work on the remaining tasks for single-material texturing.

(1) From reading through your merged PRs, it seems to me `RenderEngineVtk` and `RenderEngineGltfClient` both comply with the "short-term standard".
(2) For `RenderEngineGl`, it's using the generalized `MeshData` already, and what's left is to modify it to use `RenderMaterial`.

Can you confirm if that's the case, or do you have anything to add to help me catch up on the latest state? Thanks!

SeanCurtis-TRI, 2023-06-22:
Several things:

1. `RenderEngineVtk` is indeed up to date. (And `RenderEngineGltfClient` by inheritance.)
2. As soon as the glTF pass through merges for `RenderEngineGltfClient`, it will no longer be 100% compliant. I have a todo for addressing the very unlikely case of a gltf not having materials. (But if you have a glTF without materials, why are you using a glTF, right?)
3. I started on `RenderEngineGl` already, although I didn't get far. I'll give you a big warning now. It's not going to be pretty. Read on.

## RenderEngineGl

`RenderEngineGl` relies on a separation of concerns. It uses the concept of `OpenGlGeometry` and `OpenGlInstance`. The geometry is intended to be just what the name says, vertices, normals, texture coordinates, and triangles. It is defined in a canonical frame. `OpenGlInstance` is a tuple consisting of (pose, geometry, material). So, I can *instance* the same geometry four times, each one with a unique pose and a unique material (but there's still a single geometry).

Historically, reading OBJs was all about loading an `OpenGlGeometry`. In fact, we could specify the same OBJ multiple times and only read it once, simply instancing it for each reference. The material would come at instancing time and we'd figure out what the material is (typically using the kind of logic that is now wrapped up in the new `RenderMaterial` code).

When we read an obj/mtl combo, we're merging those two concepts. We're not *just* reading a geometry. We're reading a geometry *and a material*.  But, the thing is, we don't really have a place to store that pair. We're getting the wrong data at the wrong time in the code path.

Fortunately, things have gotten easier since I first tackled this (or, possibly, my thinking has become more clear). Let's look at where the difficulties occur:  `RenderMeshEngineGl::GetMesh()`.

When we invoke this, we *do not* pass in the geometry's `PerceptionProperties`, so when we call `LoadRenderMeshFromObj()`, it doesn't have the ability to make the full `RenderMaterial`. It will return *something* but it won't reflect the full protocol between mtl - properties - default values. Furthermore, we use the  `RenderMesh` but ignore the `RenderMaterial`.

1. We will have to pass `RenderEngineGl::GetMesh()` the geometry's `PerceptionProperties` so that `LoadRenderMeshFromObj()` can make use of them.
2. Continue to create the `OpenGlGeometry` from the returned `RenderMesh` mesh data and store it (in `RenderEngineGl::meshes_`) as is done now.
3. Now we need to track the *material*. Here's where things get *tricky*. When we see a `RenderMaterial`, we don't know if it was defined by an .mtl file, or one of the other protocols. We only see the result. If the material was defined by the `mtl` file, we *should* associate it with the geometry and use it every time that obj filename is referenced in a `Mesh` (or `Convex`). Otherwise, the material could have come from the `PerceptionProperties`, matching `.png` file, or default value. In those cases, the material *should* change from instance to instance based on a particular geometry's `PerceptionProperties`.

So, what we need, is some signal in `RenderMaterial` that communicates the *source* of the material definition. To be completely correct, it would be enough to add a `bool` member to `RenderMaterial` indicating that it came from the mesh definition. If the `RenderMaterial` came from the mesh definition, we store it somewhere in `RenderEngineGl`, mapping it to the geometry and apply it to every instance. Otherwise, we'll need to recreate the `RenderMaterial` using only properties and defaults. The good news is we have a function that does *exactly* that: `MakeMeshFallbackMaterial()` (in `render_material.h`).

The next stage would be to map `RenderMaterial` to "material" (which is really the `ShaderProgramData`). You should be able to figure out this mapping quite easily. There should be a straightforward mapping to the current logic and `RenderMaterial`'s results.

So, that was a huge treatise...some of it off the cuff as I was writing it. So, I hope it maintains some semblance of coherency. Let me know if you have any questions.

zachfang, 2023-07-13:
BTW. With the fact that #19747 has been merged, I will mark the first bullet, `RenderEngine(s) should support loading an mtl file with a single texture`, as completed.

RussTedrake, 2023-10-21:
@zachfang -- I think the following snippet is a pretty good way for you to complete the survey in the table of the OP for the mujoco parser:

```
from pydrake.all import ModelVisualizer, StartMeshcat, PackageMap

meshcat = StartMeshcat()

visualizer = ModelVisualizer(meshcat=meshcat)
visualizer.parser().package_map().AddRemote(
    package_name="mujoco_menagerie",
    params=PackageMap.RemoteParams(
        # This repository doesn't have up-to-date tags/releases; the scary
        # hash in the url is the most recent commit sha at the time of my
        # writing.
        urls=[
            f"https://github.com/google-deepmind/mujoco_menagerie/archive/94017ae66ad921978a40ac4865af49ce97d8bb75.tar.gz"
        ],
        sha256=("58e24d8718079ecbc4e94f6d8c7ca8522ec13a38ea3caaf0924d456cf6d074d5"),
        strip_prefix="mujoco_menagerie-94017ae66ad921978a40ac4865af49ce97d8bb75/",
    ),
)
visualizer.AddModels(url="package://mujoco_menagerie/kuka_iiwa_14/iiwa14.xml")
visualizer.Run(loop_once=not running_as_notebook)
```
(of course, you can look at the other example models in the menagerie, too)

I'm looking into adding texture support for the mujoco parser right now.