RobotLocomotion/drake #18859: Switching GCC debug from Focal to Jammy

jwnimmer-tri, 2023-02-23:
Recently (#18855, #18858) we've seen some failures of Debug builds that are overlooked on Focal but show up in Jammy and macOS nightly builds.

To bring those errors forward, I'd like to switch some scheduling:

In Continuous Production, please switch `linux-focal-gcc-bazel-continuous-debug` to Jammy instead.  Probably this means there's some related change in Nightly too (to align with what's present/absent from Continuous).

In pre-merge, likewise we want `linux-focal-gcc-bazel-experimental-debug` to run on Jammy instead.  I can help coordinate this with the Drake settings.

BetsyMcPhail, 2023-02-23:
`linux-focal-gcc-bazel-continuous-debug` has been removed and `linux-jammy-gcc-bazel-continuous-debug` has been added. I launched the new Continous job by hand, from now on it will launch automatically.

There are no equivalent nightly Jammy jobs, so no changes needed there.

BetsyMcPhail, 2023-02-23:
`linux-focal-gcc-bazel-experimental-debug` has been replaced with `linux-jammy-gcc-bazel-experimental-debug` as a premerge job.

jwnimmer-tri, 2023-02-23:
I've marked the new one as required in branch protections now.