RobotLocomotion/drake #18871: Upgrade orka to 2.4.0, reclaim arm64 space

svenevs, 2023-02-24:
While fixing CI issues on the backend with MacStadium they let us know a new cluster level update is available.  To complete the upgrade we need downtime to be able to rebuild the full image stack (probably a Friday is best).  There does not appear to be any imminent need to do this (cluster functionality appears to have been restored after a patch update).

- [ ] Decide on the best date to do this / schedule with mac stadium.
- [ ] While rebuilding images, exhaust any further options to [reclaim precious disk space](https://drakedevelopers.slack.com/archives/C270MN28G/p1677171647699979?thread_ts=1677159158.084489&cid=C270MN28G).

CC @jwnimmer-tri @BetsyMcPhail

jwnimmer-tri, 2023-02-28:
I've been thinking about re: disk space.

Fighting battles with 200 GiB limit on macOS is not a good use of our engineering time or budget.

I'll see if I can find a Bazel flag to make the build vastly smaller (e.g., at least 30% smaller footprint).  If not, we'll turn off some or all Debug (and ASan) builds for macOS in CI and declare it to be at-your-own-risk and unsupported to try it.

jwnimmer-tri, 2023-03-01:
Testing on a local arm mac, I see approx 155 GiB used for the build (not including externals downloads).  Of that, 30 GiB is for `//examples/...`, so if we're in a pinch one easy fix is to disable the examples in Mac Debug CI.

The syntax for that goes like this:
```
bazel test --config=clang --compilation_mode=dbg --test_timeout=300,1500,4500,-1 -- //... -//examples/...
```

Note the extra `--` to separate flags from targets, then `//... -//examples/...` to build and test everything _except_ examples.

jwnimmer-tri, 2023-03-01:
Turning off symbols ...

```
bazel test --config=clang --compilation_mode=dbg --test_timeout=300,1500,4500,-1 --copt=-g0 --host_copt=-g0 //...
```

... gets us down to 98 GiB.  That's probably the best option so far.

BetsyMcPhail, 2023-03-01:
> Turning off symbols ...
> 
> ```
> bazel test --config=clang --compilation_mode=dbg --test_timeout=300,1500,4500,-1 --copt=-g0 --host_copt=-g0 //...
> ```
> 
> ... gets us down to 98 GiB. That's probably the best option so far.

Let's try it, I'll make a drake-ci PR and see how it goes.

jwnimmer-tri, 2023-03-01:
Fine by me.  To be clear, we do not want this in Ubuntu Debug.

I have a snapshot of file sizes for the existing full Debug build.  I'm also going to see if I can find a slightly less brutal fix, perhaps changing how we link all of our unit tests.

jwnimmer-tri, 2023-05-22:
Given https://github.com/RobotLocomotion/drake-ci/pull/205, I think I'd be happy to just close this one as "good enough".  Is there anything more we need to do here?

>  I'm also going to see if I can find a slightly less brutal fix, perhaps changing how we link all of our unit tests.

I ran out of time before I was able to find anything.

jwnimmer-tri, 2023-10-05:
@BetsyMcPhail Do we still need to upgrade orka to 2.4.0?

svenevs, 2023-10-05:
No, we're up to date at this point.

closed 2023-10-05 by jwnimmer-tri