RobotLocomotion/drake #18878: AddModelsFromString throws, but AddModelsFromUrl does not

RussTedrake, 2023-02-27:
### What happened?

On this simple `table.sdf`:
```
<?xml version="1.0" ?>
<sdf version="1.9">
  <world name="table_world">
    <model name="table">
      <static>true</static>
      <pose>0.0 0.0 -0.005 0 0 0</pose>
      <link name="table_link">
	<collision name="table_collision">
	  <geometry>
	    <box><size>1. 1. 0.001</size></box>
	  </geometry>
	</collision>
	<visual name="table_visual">
	  <geometry>
	    <box> <size>2. 2. 0.001</size> </box>
	  </geometry>
	  <material>
	    <ambient>0.8 0.8 0.8 1</ambient>
	  </material>
	</visual>
      </link>
    </model>
  </world>
</sdf>
```
I was able to parse it successfully using 
```
parser.AddModelsFromUrl("package://QR/grape_quest/table.sdf")
```
but calling 
```
model_directives = f"""
directives:
- add_model:
    name: {name}
    file: package://QR/grape_quest/table.sdf
"""
parser.AddModelsFromString(model_directives, 'dmd.yaml')
```
resulted in the parser throwing
```
RuntimeError: File must have a single <model> element.
```
I was surprised by the difference!  (and expected both to work, given my quick re-read of the sdformat docs).

### Version

master

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

rpoyner-tri, 2023-02-27:
I agree this state of affairs is unsatisfactory, and it exposes a corner of #18059 that hadn't yet reached my attention. The dmd.yaml statements are essentially serializations of calls to the Parser and Mbp public apis. So, if I'm migrating the Parser API, I should also be doing the analogous thing in dmd.yaml. The existing dmd.yaml `add_model` calls Parser.AddModel() (singular), which in turn traces in to the SDFormat parser via AddModelFromSdf(), which does indeed enforce a single-model rule.

There are a couple of ways this could go:
(1) implement `add_models` (plural) in dmd.yaml, and eventually deprecate `add_model`.
(2) just broaden the semantics and documentation of `add_model` so that it always routes through a Parser plural method, and hence drops the single-model rule enforcement. Opinions welcome as to whether this would be a breaking change.

rpoyner-tri, 2023-02-27:
In either of the above implementation routes, it probably makes sense to expose some the Parser API modernizations as they come on line -- prefixing (via Parser prefix constructor), auto renaming (still in dev branches), etc.

jwnimmer-tri, 2023-02-27:
Does `table.sdf` actually induce multiple model instances, though?  Possibly it's a bug that it does so?  That seems like the crux of the issue here.  The parser plumbing is secondary.

In any case, as far as dmd semantics -- one option would be to remove the validity check that `add_model` must always specify a `name`.  When no name is specified, it could use the plural parser.

rpoyner-tri, 2023-02-27:
The supplied sdf is using `<world>` format, but only has a single model; arguably that points to a bug in the checking in AddModelFromSdf().

RussTedrake, 2023-02-27:
I agree.  When I looked at the parser code last night, it looked like it automatically said "no unique root" if both a world and model were present, which seems like a bug to me.  (but I've not studied the sdf world element semantics).  If we agree, then I think fixing that issue would be the first order of business -- and would get me unstuck in this particular use case.

rpoyner-tri, 2023-03-02:
I beleive #18888 has fixed the immediate symptom. I'll move my reservations about dmd.yaml statements over to #18059 for now.

closed 2023-03-02 by rpoyner-tri