RobotLocomotion/drake #18902: CI failures trying to fetch rules_cc

BetsyMcPhail, 2023-02-28:
### What happened?

https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-continuous-everything-leak-sanitizer/1679
https://drake-jenkins.csail.mit.edu/job/linux-focal-clang-bazel-experimental-release/5484
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-experimental-release/2663

### Relevant log output

```shell
WARNING: Download from https://github.com/bazelbuild/rules_cc/releases/download/0.0.2/rules_cc-0.0.2.tar.gz failed: class java.io.FileNotFoundException GET returned 404 Not Found
ERROR: An error occurred during the fetch of repository 'rules_cc':
```

BetsyMcPhail, 2023-02-28:
Slack conversation [here](https://drakedevelopers.slack.com/archives/C270MN28G/p1677596598083889) and [here](https://drakedevelopers.slack.com/archives/C270MN28G/p1677531122453029)

jwnimmer-tri, 2023-10-05:
Summary from slack, for posterity:

> Jeremy: Given some of the errors in the past couple of days, it seems like perhaps rules_cc is now an implicitly defaulted external that bazel is fetching on its own from github.  If that's true, then we should be explicit about it -- add it to our workspace and mirror it ourselves.

So the request here is simple ... add `rules_cc` to `tools/workspace/...` stuff, so that we get mirroring.

jwnimmer-tri, 2024-11-01:
Fixed in #21545.

closed 2024-11-01 by jwnimmer-tri