RobotLocomotion/drake #18913: Hydroelastic user guide features unsupported drake_visualizer

SeanCurtis-TRI, 2023-03-01:
### What happened?

The user guide was written when `drake_visualizer` was still supported (and represented the "premium" hydro visualization mechanism). Those days are gone and it we should no longer feature that in the user guide. The discussion and images should be updated to reflect the current set of tools and visualization capabilities.

In addition to visual quality that hasn't currently been reproduced, we go so far as to show `drake_visualizer`-only visualizer configuration. E.g.: 

[image]

See the guide's current treatment on [Visualizing hydroelastic contact](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html#hug_visualizing).

### Version

v1.13.0

jwnimmer-tri, 2023-07-29:
@DamrongGuoy @rpoyner-tri This becomes ever more pressing, as almost everyone is using Jammy now and Focal support will be dropped in ~7 months.

It also relates to #16216, where several hydro examples are still hard-coded to use `//tools:drake_visualizer`.

DamrongGuoy, 2023-08-01:
@jwnimmer-tri Thanks for asking.  Please kindly have `#drake-adoption` work on it.

jwnimmer-tri, 2023-09-12:
> There's no mention of introducing compliant meshes (as vtk file). This is obviously something that would likewise be worthwhile to note in the user guide.

That's a pretty distinct request vs fixing the visualization, so I've broken it out into #20176.  I'll leave this ticket focused on fixing our documentation related to hydro visualization.

PR #20175 (2023-09-13): [doc] Refresh hydroelastic guide instructions for visualization
In support of that, update the `ball_plate` demo to use Meldis.

Closes #18913.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20175)
closed 2023-09-13 by jwnimmer-tri