RobotLocomotion/drake #18917: Model visualizer should know about mimic tags (coupled joints)

jwnimmer-tri, 2023-03-01:
See #17663 and #18728 for background.

When we are displaying a model with coupled joints, we should be enforcing the constraint during visualization.

jwnimmer-tri, 2023-03-08:
I'd like to have a separate issue for the inverse dynamics controller.  Could you please file one explaining what's happening, with a high-level outline of how to reproduce the problem, and what you'd prefer to happen vs what actually happens?

I think one thing we lack is any robot model anywhere in Drake that uses mimic joints.  We should work to have at least one in-tree, so that the all of us (including the dynamics team) can more easily understand and reproduce problems like this.  We can collaborate across the two (or more) mimic-related issues to get the model added -- whichever one gets there first can add the file(s).  (Possibly "no example of mimic joints" is documentation-like defect all on its own, and we should open a new issue for it, and coordinate there.)

I suspect when we throw it at other subsystems within Drake it will likewise fail to perform as desired.

My reasoning for using separate issues is that different people will probably work on them, with different priorities.  To my understanding at least, the lack of effective mimic joints in model_visualizer is embarrassing (and maybe should have a warning?) but is not a major roadblock to collaboration.  I expect that things like IDC failures are much more of a blocker, so we should prioritize them more urgently.

rpoyner-tri, 2023-05-22:
FWIW, I don't think this ticket has any strong connection to "mimic tags" -- won't the visualizers be reading constraints directly from the model(s) in Mbp?

MengMengMengH, 2024-04-03:
> The limitation is more general than model visualizer. For instance, the inverse dynamics controller currently also falls down if a model is specified with constraints (e.g. elements in urdf). This limitation popped up on our MIT project with Movo, which uses tags.`mimic``mimic`
> 
> I think the right solution is to use a general machinery like rather than some narrow solution for model visualizer. That would suggest we might want to generalize the title of this issue and keep them in one place? I'm also ok if we want a separate issue to track.`SystemConstraints`
> 
> Also, if 2 of our university friends are both running into this at the very beginning of their projects, perhaps it deserves a higher priority?

I met the same problem here.When I want to use <InverseDynamicsControllers> in my simulation,I was told that it was a continous system and would ignore the drake:mimic joint.
I wonder if there are any solutions. I would be glad to receive your reply.