RobotLocomotion/drake #18930: More informative error messages when loading YAML files

dmcconachie-tri, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
I am frequently frustrated when loading from YAML files and there is a schema mismatch somewhere. For example:
```
what():  YAML node of type Mapping (with size 4 and keys {foo,bar}) is missing entry for SubSchema foo while accepting YAML node of type Mapping (with size 4 and keys {bar}) while visiting SomeSchema sub_schema.
```

**Describe the solution you'd like**
I would like 2 things:
1. The error message tells me what file is being loaded when this happened.
2. The error message tells me the parse chain where this happened. I.e.; the list of parent nodes traversed to get to this bad node.

**Describe alternatives you've considered**
For part 1; I've sprinkled
```c++
drake::log()->info("Loading schema <identifier I will recognize> from [{}]", file);
```
all over the place so that I see that immediately before the exception text.
For part 2: I don't have a good solution, though often it isn't necessary. Except when it is needed it's often because there are 10+ candidates to look through, once I have the right file.

**Additional context**
The call signature I am using is:
```c++
drake::yaml::LoadYamlFile<Schema>(filename, std::nullopt /* no child name */, Schema{} /* default schema */);
```

FYI @ggould-tri per our Slack conversation.

jwnimmer-tri, 2023-03-02:
To help us make the right fix, please provide an example schema file and malformed data file.

dmcconachie-tri, 2023-03-02:
Sure, no problem. Might not do so today.

jwnimmer-tri, 2023-03-06:
I did a 10 minute prototype [here](https://github.com/jwnimmer-tri/drake/tree/yaml-error-details).  It seems like we should be able to print the filename, line number, and column number for any error.

dmcconachie-tri, 2023-03-06:
Example:
```C++
#include <gtest/gtest.h>

#include <cstdlib>
#include <fstream>
#include <string>

#include "drake/common/yaml/yaml_io.h"

namespace drake {
namespace yaml {
namespace {

struct BottomLevel {
  template <typename Archive>
  void Serialize(Archive* a) {
    a->Visit(DRAKE_NVP(int1));
    a->Visit(DRAKE_NVP(int2));
  }
  int int1{};
  int int2{};
};

struct MidLevel {
  template <typename Archive>
  void Serialize(Archive* a) {
    a->Visit(DRAKE_NVP(bottom));
  }
  BottomLevel bottom;
};

struct TopLevel {
  template <typename Archive>
  void Serialize(Archive* a) {
    a->Visit(DRAKE_NVP(middle));
  }
  MidLevel middle;
};

GTEST_TEST(YamlErrorMessages, ExampleTest) {
  const char* test_tmpdir = std::getenv("TEST_TMPDIR");
  const std::string filename = test_tmpdir + std::string("/schema.yaml");
  std::ofstream file(filename);
  file << "middle:\n";
  file << "  bottom:\n";
  file << "    int1: 1\n";
  file.close();
  LoadYamlFile<TopLevel>(filename);
}

}  // namespace
}  // namespace yaml
}  // namespace drake
```
yields
```
C++ exception with description "YAML node of type Mapping (with size 1 and keys {int1}) is missing entry for int int2 while accepting YAML node of type Mapping (with size 1 and keys {bottom}) while visiting drake::yaml::(anonymous)::BottomLevel bottom while accepting YAML node of type Mapping (with size 1 and keys {middle}) while visiting drake::yaml::(anonymous)::MidLevel middle." thrown in the test body.
```

I'll test your prototype shortly.

jwnimmer-tri, 2023-03-06:
Hah, no need to test the prototype yet.  It only has line numbers.  I'll ping you once I get the filenames in there.

jwnimmer-tri, 2023-03-06:
When reporting an error ...
Is the first line of the file line 0 or line 1?
Is the first character of a line column 0 or column 1?
GCC seems to use 1 for both, so I'll do the same.

dmcconachie-tri, 2023-03-07:
Agreed, matches the vim "goto line number" command too. Presumably emacs as well.

jwnimmer-tri, 2023-03-07:
Emacs is line 1, column 0.

> In Column Number mode, the displayed column number counts from zero starting at the left margin of the window. If you would prefer for the displayed column number to count from one, you may set `column-number-indicator-zero-based` to `nil`.

However, compilation-mode knows that compilers start from column 1, so `(next-error)` still goes to the right place.

PR #18952 (2023-03-16): [yaml] Add filename, line, and column to error messages
Closes #18930.

+@dmcconachie-tri you get right of first refusal for feature review.  Let me know if you'd like to defer.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18952)
closed 2023-03-16 by jwnimmer-tri