RobotLocomotion/drake #18937: DirectCollocation stores the context for each waypoint

hongkai-dai, 2023-03-04:
Currently `DirectCollocation` class stores a single context, and re-use that context for all waypoints, when imposing the dynamics constraints. The cache stored in that context is wiped out, when DirectCollocation moves to impose the dynamics constraints for the next waypoint. But that context is very useful, if we later want to impose some constraint that can re-use the cache (for example a kinematic constraint). Here is a use case
```py
dircol = DirectCollocation(diagram, context, ...) # Impose the collocation constraint for all the waypoints.

# Now I want to impose some kinematic constraint on one waypoint, but I will need to re-compute the cache.
diagram_context_waypoint = diagram.CreateDefaultContext()
plant_context_waypoint = plant.GetMyMutableContextFromRoot(diagram_context_waypoint)
dircol.prog.AddConstraint(PositionConstraint(plant, ..., plant_context_waypoint), dircol.state(waypoint_index)[:plant.num_positions()]))
```

I think if DirectCollocation class stores a vector of contexts, then we could re-use the cache in that context for imposing the kinematic constraint.
```cc
class DirectCollocation {
  private:
    std::vector<std::unique_ptr<systems::Context<double>>> waypoints_context_;
};
```

What do you think? @RussTedrake

RussTedrake, 2023-03-05:
I completely agree.  We've had a todo about it since the times when caching was still a proposal, but didn't exist yet.
https://github.com/RobotLocomotion/drake/blob/42d86973c41725ad4ef1848547374fa21263009f/planning/trajectory_optimization/direct_collocation.cc#L126
I'd love to finally improve it.

RussTedrake, 2023-03-16:
fwiw -- i implemented this this morning (while prepping for my trajopt lecture).  will polish and PR it shortly.

PR #19019 (2023-03-22): Implements caching in DirectCollocation.
Resolves #18937.

+@hongkai-dai for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19019)
commit 134f6575 (2023-03-22): Implements caching in DirectCollocation. (#19019)

Resolves #18937.
closed 2023-03-22 by RussTedrake