RobotLocomotion/drake #18940: Mosek optimization

ZZHPKU, 2023-03-05:
### What happened?

I add a quadratic unequal constraint through prog.AddConstraint() and the problem can be solved by SNOPT. However, when I solved it by MosekSolver, a error "MosekSolver is unable to solve because a GenericConstraint was declared but is not supported." appeared. Hoe can I use MosekSolver?

hongkai-dai, 2023-03-05:
May I ask how you add the quadratic inequality constraint into MathematicalProgram? Currently Drake's MathematicalProgram intentionally doesn't support adding a quadratic inequality constraint and then invoke Mosek solver. The reason is that Mosek suggests not to impose a quadratic inequality constraint, but instead impose a Lorentz cone constraint. Namely if you have a quadratic constraint
```
x' * A' * A * x + 2*(A*b)'*x <= c
```
which is equivalent to
```
norm(A* x + b) <= c - b'*b
```
then impose it as
```
[sqrt(c-b'*b)] in the Lorentz cone.
[  A*x + b   ]
```

I guess your constraint is probably not parameterized in the formulation above, but instead like
```
x' * Q * x + p' * x <= t
```
namely you have not done a decomposition of your `Q` matrix to `Q = A'*A` yet. This quadratic constraint is equivalent to a Lorentz cone constraint
```
t + 1/4 * p'*Q⁻¹ * p >= |A*x + 1/2*A⁻ᵀ*p|²
```
You can call
```cc
// Compute the term 1/4 * p'*Q⁻¹ * p
Eigen::SelfAdjointEigenSolver es(Q);
const Eigen::VectorXd Q_inv_p = es.solve(p);
const double lhs = std::sqrt(t + 0.25 * p.dot(Q_inv_p));
prog.AddLorentzConeConstraint(lhs, x.dot(Q * x) + p.dot(x) + 0.25 * p.dot(Q_inv_p));
```

Later I can add a function in `MathematicalProgram` to make this process easier, but for the moment you can use the code snippet above.

ZZHPKU, 2023-03-06:
Yea, I just add it in the  form of  x' * Q * x + p' * x <= t.
LorentzCone works well, thank you!

closed 2023-03-06 by hongkai-dai