RobotLocomotion/drake #18945: Make debugging easier when MathematicalProgram's constraints contain inconsistent bounds

hongkai-dai, 2023-03-06:
If we have a MathematicalProgram whose bounds are in-consistent, then a lot of solvers will abort directly. It is quite hard to debug this solver failure for many users.

The constraints bounds can be in-consistent in two ways:
1. We have one constraint whose lower bound is larger than the upper bound.
2. We have several BoundingBoxConstraint imposed on the same variable, and even within each BoundingBoxConstraint its lower bound is smaller than the upper bound, the lower bound of BoundingBoxConstraint `bb_con1` is larger than the upper bound of another BoundingBoxConstraint `bb_con2`, and hence when we aggregate all the bounding box constraints, the resultant lower bound is larger than the upper bound. Here is one example
   ```
   prog.AddBoundingBoxConstraint(1, 3, x);
   prog.AddBoundingBoxConstraint(-1, 0, x);
   // The aggregated constraint is 1 <= x <= 0, the lower bound is larger than the upper bound.
   ```

Currently when the solver aborts, the user receives an error code, hidden in `result.get_solver_details()`. Many users don't check this error code. And even if they do, still that error code only says "91" (in SNOPT), or some other integer in Mosek. The user need to check the solver manual, and then find that 91 means "problem input data error" in SNOPT. Even after all these efforts, the user still don't know what is the exact input data error.

I would propose to return more debugging information to the user to help them locate the error in the constraint bounds.

hongkai-dai, 2023-03-06:
@jwnimmer-tri made some suggestions in https://github.com/RobotLocomotion/drake/pull/18936#issuecomment-1456279560, I paste them here

One answer might be trajectory-optimization-specific sugar that makes this error impossible (an exception) in the first place, instead of trying to fix it at the lower-level solvers where things are more murky.

My next question is whether SNOPT's "Print file" output contained any hints about this failure. If it did, can we update the tutorial to better highlight how to use that info? Why didn't the user find it on his own?

Assuming that the SNOPT log doesn't help, then it's up to us to guide the user.

Here's what the tutorial currently shows to help debug this:
```
infeasible_constraints = result.GetInfeasibleConstraints(prog)
for c in infeasible_constraints:
    print(f"infeasible constraint: {c}")
```
I think your premise is that's not good enough? Maybe because too many of the constraints were infeasible, so the lower > upper detail on one of them gets lost in the noise? Or did the user not even try it?

The general thrust of my proposal would be to expand that section of the tutorial, adding more functions if need be.

The first point would be that in GetInfeasibleConstraints, if we have lower > upper, then we should use a different description string, something like adding scream caps "ALWAYS FALSE" into the message. We could also sort these to the front of the list of returned strings (i.e., prioritize the advice).

If we find that's still not enough, we could imagine adding more kind of help strings to that vector, or even creating a second function like std::string HelpMeDiagnoseUnsolveableProgram() const; that returns whole paragraphs of advice and hyperlinks, particular to the program in question.

My bottom line is that:

(1) We use a tutorial-centric approach. If the user isn't reading the existing tutorial that already helps them, fix that first. Don't add more features when the users aren't even using the current ones.

(2) We use a result-centric approach. During program setup and solve, we shouldn't pay the cost of trying to anticipate debugging questions; we should run the solve quickly, and without console spam. If there is an interactive user who is struggling, then they can call functions that help guide them through debugging, given a failed MathematicalProgramResult. The tutorials will show them how. We could even imagine that to_string(MPResult) contains 1 hyperlink, to the tutorial or a Doxygen overview that elucidates how to debug.

hongkai-dai, 2023-03-06:
Here are my thoughts on Jeremy's questions:

> One answer might be trajectory-optimization-specific sugar that makes this error impossible (an exception) in the first place, instead of trying to fix it at the lower-level solvers where things are more murky.

The problem of inconsistent constraint bounds can appear in any mathematical optimization, not just in trajectory optimization. So I don't think we should fix in in traj-opt.

> My next question is whether SNOPT's "Print file" output contained any hints about this failure. If it did, can we update the tutorial to better highlight how to use that info? Why didn't the user find it on his own?

Unfortunately SNOPT's print file doesn't hint about the failure. Beside the print out file can be very big so hard for the user to find the hint message out from the big file.

> Here's what the tutorial currently shows to help debug this:
> ```
> infeasible_constraints = result.GetInfeasibleConstraints(prog)
> for c in infeasible_constraints:
>   print(f"infeasible constraint: [...]

I don't think this will work. What that `GetInfeasibleConstraint` does is to take the solution returned from SNOPT, evaluate each constraint, and then return the constraint whose evaluation at the SNOPT solution is outside of the bounds. In this case, SNOPT returns the initial guess directly. If we evaluate all the constraints at the initial guess, it probably will violate all constraints, and it is really hard for the user to find out the one with inconsistent bounds. What make it even worse is when we have two BoundingBoxConstraint with one `bb_con1` having larger lower bound than `bb_con2` upper bound. Consider this example
```cc
prog.AddBoundingBoxConstraint(1, 3, x);
prog.AddBoundingBoxConstraint(-2, 0, x);
// Now solve with initial guess x= -1
result = Solve(prog, Vector1d(-1));
result.GetInfeasibleConstraints();
```
In this case it will return the infeasible constraint as 1 <= x <= 3, which by itself there is nothing wrong with this constraint (the lower bound of this constraint is smaller than its upper bound), so the user will be confused.

jwnimmer-tri, 2023-03-06:
To be clear, in your above you say the solver "aborts", but I don't think you mean that in the conventional sense of aborting the process.  I think you just mean that it returns a `MathematicalProgramResult` saying that the program could not be solved.  Something like "immediately gives up" rather than "aborts".

> I don't think this will work.

I think my point still holds.  In the `GetInfeasibleConstraint` code, we can search for vacuously-false constraints (or pairs of bbox constraints).  If there are any, we can return a specific string for them and we can make it the _first_ string in the vector of results, and it can be in scream caps:

```c++
std::vector<std::string> descriptions;
descriptions.emplace(fmt::format(
    "The constraint {} is ALWAYS FALSE because its bound lower {} is greater than its upper bound {}.",
    ... ));
descriptions.emplace(fmt::format(
    "Together, the constraints {} and {} are ALWAYS FALSE because their bounds ({}, {}) and ({}, {}) "
    "for decision variable {} are non-overlapping.",
    ... ));
```

In short -- whatever you wanted to warn during prog.AddConstraint, we can put into the description string that we've told users to look at when they are debugging.

hongkai-dai, 2023-03-06:
> To be clear, in your above you say the solver "aborts", but I don't think you mean that in the conventional sense of aborting the process. I think you just mean that it returns a [...]

That is right. I mean "immediately gives up".

I like the idea of improving `GetInfeasibleConstraints`, and we can return constraints with inconsistent bounds here. We will need to modify the API of `GetInfeasibleCnstraints` to also return a string `description`. What would be the easy way to make this change? If I add `std::string` to the return of `GetInfeasibleConstraints` as
```cc
std::pair<std<vector<Binding<Constraint>>, std::string> GetInfeasibleConstraints(const MathematicalProgram& prog, std::optional<double> toleance);
```
it will break the current code, as C++ does not overload based on the return type. This new `GetInfeasibleConsraints` will conflict with the current version that only returns `std::vector<Binding<Constraint>>`

jwnimmer-tri, 2023-03-06:
Sorry, I see the disconnect now.  I was saying the wrong function name!

I meant that we should change `GetInfeasibleConstraintNames`, which already returns strings.