RobotLocomotion/drake #18946: Upgrade to Bazel 6.1

jwnimmer-tri, 2023-03-06:
We should upgrade our supported Bazel version to https://github.com/bazelbuild/bazel/releases/tag/6.1.0.

FYI the prior upgrade was here: https://github.com/RobotLocomotion/drake/issues/18508.

It might be worth planning to upgrade Ubuntu and macOS separately.

ggould-tri, 2023-03-14:
AIUI we're expecting 6.1.1 in the next day or two?  I don't quite see how the bazel release schedule works.
Anyway, obligatory bazel release link for reference: https://github.com/bazelbuild/bazel/milestone/49?closed=1

jwnimmer-tri, 2023-03-14:
FYI https://github.com/bazelbuild/bazel/issues/17212#issuecomment-1467021040 for updates.

closed 2023-03-16 by jwnimmer-tri

BetsyMcPhail, 2023-03-17:
Provisioned Linux images have been updated (Mac does not need to be updated)

PR #19012 (2023-03-16): Upgrade to Bazel 6.1.1
Resolves #18946  

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19012)