RobotLocomotion/drake #18951: Ball joint shouldn't report the correct error when there is an axis.

marcoag, 2023-03-07:
### What happened?

Currently, when parsing ball joints, there is no warning recorded if there is an axis. Instead of the warning, the parser goes ahead and throws a limits related warning that doesn't seem very relevant for a ball joint that shouldn't have axis (see [here](https://github.com/RobotLocomotion/drake/blob/153f3fb18b8c2f5203ffb0531e96af75610c5040/multibody/parsing/detail_sdf_parser.cc#L464)).

We should correct this and add a warning similar to the one that is there for [axis2 in ball joints.](https://github.com/RobotLocomotion/drake/blob/153f3fb18b8c2f5203ffb0531e96af75610c5040/multibody/parsing/detail_sdf_parser.cc#L472)

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

rpoyner-tri, 2023-03-07:
@marcoag you appear to have the context. Are you willing to take this?

marcoag, 2023-03-07:
Please do. I will follow up with a PR on this.

scpeters, 2023-03-20:
Here's my thoughts:

Every physical mechanism has friction of some kind, including ball joints. In SDFormat we currently store this information in the `//joint/axis` and `//joint/axis2` elements, which allows different friction parameters for each axis of a multi-axis joint. It is a little awkward for ball joints, to use the `//axis` element because `//axis/xyz` is required but doesn't have meaning for a ball joint. I think the simplest thing would be to just store the information in `//joint/axis/dynamics/friction` for a ball joint and ignore the `//joint/axis/xyz` element, or explicitly set it to `0 0 0`.

With this approach, the `//joint/axis/dynamics` tag would be relevant for ball joints, and the `//joint/axis/xyz` will just be there since it's a required element, but it could be ignored or set to `0 0 0`.

In terms of interpreting these values, it's important to be clear about it. I would interpret the friction and damping values in `//axis/dynamics` as opposing any relative angular motion. I think I would ignore the spring stiffness and reference parameters.

sherm1, 2023-03-20:
Thanks, Steve! 

That seems like a workable hack, but is clearly a bug in the sdf definition ("axis" makes NO sense for a ball joint!). Is there a way to get this fixed in a future sdf spec update?

jwnimmer-tri, 2023-04-01:
Something else we should probably consider in the schema design here ...

Do we ever want to support _actuated_ ball joints?  If so, then we'll need a way to specify that, along with an effort limit.  Similarly for other things in `<limits>` like if we want to place a bounding box on the rpy angles, or velocities, etc.

sherm1, 2023-04-01:
IMO it makes no sense to talk about actuated ball joints since AFAIK there is no physical mechanism for that. A similar effect can be produced with a 3d _gimbal_ joint (i.e. orientation modeled with roll-pitch-yaw), or a series of 3 orthogonal revolute joints. It is possible to build a real system like that.

jwnimmer-tri, 2023-05-17:
BTW This issue has been about SDFormat files.  It's probably worth looking whether the URDF parser has the same hazard, and if so opening another issue.

closed 2023-05-24 by rpoyner-tri