RobotLocomotion/drake #18968: Documentation for wheel PR local testing

jwnimmer-tri, 2023-03-09:
If a code reviewer on a wheel-related PR would like to test it in `pip` locally, there's no easy / clear instructions for this.  We should write some.  Probably the new doc would live in the `tools/wheel/README`.

The idea would be to test using the experimental wheels built from CI, rather than building locally.

General gist:
* Offer a copy-paste command(s) for the jenkins bot to run some (or all) wheel builds on a PR.
* Explain how to find the whl download links in the PR output.
* Explain how to install and run it locally for testing (basically https://drake.mit.edu/pip.html#nightly-releases but with a different URL).

The primary focus would be for running-on-Jammy developers (manylinux cp310).  It's fine to document other platforms also, but the most important thing is for the Jammy workflow to be clear and obvious.

svenevs, 2023-04-24:
Users will need to be informed of what to install, see https://github.com/RobotLocomotion/drake/issues/19264#issuecomment-1520703895

jwnimmer-tri, 2023-04-24:
FYI for myself locally, this is what I currently have installed:
```
ii  docker.io      20.10.21-0ubuntu1~22.04.2 amd64        Linux container runtime
```

jwnimmer-tri, 2023-10-05:
For the record, Ubuntu has temporarily broken the buildkit support in their packages.

https://bugs.launchpad.net/ubuntu/+source/docker-buildx/+bug/1968035

I worked around it by installing `docker-buildx=0.11.2-0ubuntu1~22.04.1` from `jammy-proposed` for the moment.  Hopefully that makes its way into mainline Ubuntu soon enough.  (Monday would be the earliest, with their 7-day minimum in `jammy-proposed`.)

mwoehlke-kitware, 2023-11-02:
`tools/wheel/README[.rst]` a) exists, and b) documents how to build wheels. IOW, it's fairly low-level documentation. It seems that this is asking for higher level documentation, maybe even belonging in or near https://drake.mit.edu/developers.html?

For the first bullet:
```
curl https://raw.githubusercontent.com/RobotLocomotion/drake/jenkins-jobs-experimental/request-jobs-unprovisioned.txt | grep wheel
```

jwnimmer-tri, 2023-11-02:
The goal is to allow reviewers to run the PR changes, but without running a local build.

The general sequence we'd like to document all in once place would be something like (a) how to run the bot command (b) how to find the whl URL (much easier now!) then (c) how to 'pip install' that URL into a venv.

jwnimmer-tri, 2023-12-12:
> Am I missing something?

Nope, looks like it probably got a bit better on its own after the ticket was filed.  The PR is a fine conclusion for this ticket.

PR #20661 (2023-12-13): Improve wheel testing instructions
Update the instructions for how to find the URL of a wheel built as part of a pull request experimental build (the links are now easier to find). Add instructions for how to locally install the wheel.

Fixes #18968.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20661)
closed 2023-12-13 by xuchenhan-tri