RobotLocomotion/drake #18971: [geometry, sensors] Possible to expose image loading functionality publicly?

EricCousineau-TRI, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
See discussion here re: loading images:
https://github.com/RobotLocomotion/drake-ros/pull/133#pullrequestreview-1334953151

**Describe the solution you'd like**
If possible, having a public view into code that has equivalent functionality and robustness as:
https://github.com/RobotLocomotion/drake/blob/v1.13.0/geometry/render_gltf_client/internal_render_client.cc#L322-L385

**Describe alternatives you've considered**
We considered using a third-party library for C++, such as `lodepng`. But for things that depend on `drake`, it'd be nice to avoid needing third-party libs if we already have a "conduit" to that functionality through Drake itself.

**Additional context**
Ya hydroelastic!

Assigning Zach as component, wasn't sure if it should be @joemasterjohn based on illustration though
\cc @jwnimmer-tri

jwnimmer-tri, 2023-03-13:
FYI I toyed around with an idea for this a couple of years ago (for both loading and saving):

https://github.com/jwnimmer-tri/drake/commits/image-io

For the loading side, we'd need to promote the diagnostic policy to a public API.  Any image-loading function needs to be able to report structured errors, and it'll be a while before we have access to [std::expected](https://en.cppreference.com/w/cpp/utility/expected).

jwnimmer-tri, 2024-02-13:
Fixed by #20447 in v1.24.0.

At the moment, errors are communicated by throwing exceptions.  Some kinds of errors can be detected ahead of time by calling `LoadMetadata` first.

closed 2024-02-13 by jwnimmer-tri