RobotLocomotion/drake #18977: ModelVisualizer.AddModelsFromUrl()

RussTedrake, 2023-03-11:
Now that we are recommending that people use packages to identify their files, I *believe* we should provide `ModelVisualizer.AddModelsFromUrl()` in addition to `AddModels()`.

jwnimmer-tri, 2023-03-11:
I agree.

For the moment, the workaround is `vis.parser().AddModelsFromUrl(...)` with the downside that the reload button would no longer be available.

As far as the function signature, another option is `AddModels(url=...)`, i.e., using a kwarg instead of a separate function.

RussTedrake, 2023-03-12:
If we do the kwarg here, we should do it in `Parser`, too.  There are a lot of variants floating around!

jwnimmer-tri, 2023-03-13:
The difference is that the Parser written in C++, where it's not possible to use named kwargs.

Perhaps the Python Parser API could diverge from the C++ Parser API, but so far we've been hesitant to do that, under the premise that translating a Python tutorial back down into C++ should be rote (e.g., changing method-call-dots to member-function-call-arrows), rather than altering function names.

jwnimmer-tri, 2023-04-05:
I ended up need this for making slide deck videos, so I went ahead and implemented it.  See #19131.

PR #19131 (2023-04-06): [visualization] Enhance model_visualizer command line
Closes #18977.

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19131)
closed 2023-04-06 by xuchenhan-tri