RobotLocomotion/drake #18980: ManipulationStation doesn't appear in doxygen

RussTedrake, 2023-03-12:
### What happened?

I went to look at the ManipulationStation documentation, but it seems to be missing.  Only the ManipulationStationHardwareInterface shows up.

[image]

jwnimmer-tri, 2023-03-13:
Possibly the unnecessary forward declaration in #18268 is what did it?

jwnimmer-tri, 2023-03-14:
+@amcastro-tri as feature reviewer on #18268, could I ask you to investigate this and see if that commit caused this regression, and maybe fix it?

amcastro-tri, 2023-03-15:
No problem, let me try

jwnimmer-tri, 2023-03-21:
I'll take a look.

jwnimmer-tri, 2023-03-21:
I passed `--verbose` to the doc build command, and immediately saw that the `system_yaml_to_html` function was crashing:

```
  File "/tmp/doc_builder_temp_qxn79d0y/drake/doc/doxygen_cxx/system_doxygen.py", line 38, in system_yaml_to_html
    if "input_ports" in y:
TypeError: argument of type 'NoneType' is not iterable
```

The proximate cause for that is the _other_ thing obviously wrong with the PR (that I didn't mention before, because I didn't see any way it could be related here; oops!) -- the comment opener on the new class's overview uses `//` instead of `///`.  If you change it to `///` the Doxygen shows up fine.

Can you work on landing that fix?

I'll working on trying to fix Doxygen to give better feedback in this situation.

closed 2023-03-21 by jwnimmer-tri

amcastro-tri, 2023-03-21:
wow, nice job @jwnimmer-tri! thank you!

PR #19004 (2023-03-16): Removes unnecessary forward declaration from ManipulationStation
Remove in the attempt to fix #18980. It does not solve it, but definitely an unnecessary change.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19004)
commit 015b2f97 (2023-03-16): Removes unnecessary forward declaration from the header file. (#19004)

Removes forward declaration accidentally introduced in #18268 in manipulation_station.h.
We thought this would fix #18980 (ManipulationStation does not show up in Doxygen) but it did not.
PR #19041 (2023-03-21): [doc] Reduce the fallout from system_html crashes
Closes #18980.

+@ggould-tri for both reviews, please.

\CC @amcastro-tri FYI

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19041)