RobotLocomotion/drake #18981: Drake tools is failing on the ubuntu 22.04

mamurtaza88, 2023-03-13:
I am trying to build drake from source on ubuntu 22.04. My laptop is alienware. I followed all the commands on the website. However, it crashes while doing the command
```
bazel build \\...
```
This result in terminal tab being closed and system hanged. In addition, if I do the following command
```
 bazel run //tools:meldis -- -w &
```

I get the following message
```
bazel run //tools:meldis -- -w &
[1] 97800
murtaza@murtaza-Alienware-m17-R3:~/drake$ Starting local Bazel server and connecting to it...
INFO: Analyzed target //tools:meldis (207 packages loaded, 12769 targets configured).
INFO: Found 1 target...
INFO: Deleting stale sandbox base /home/murtaza/.cache/bazel/_bazel_murtaza/5e9d8839be9455185d24b5bb992beee6/sandbox
ERROR: /home/murtaza/drake/bindings/pydrake/systems/BUILD.bazel:46:21: Compiling bindings/pydrake/systems/framework_py_systems.cc failed: (Exit 1): gcc failed: error executing command (from target //bindings/pydrake/systems:framework.cpython-310-x86_64-linux-gnu.so) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 233 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
gcc: fatal error: Killed signal terminated program cc1plus
compilation terminated.
Target //bindings/pydrake/visualization:meldis failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 105.407s, Critical Path: 97.90s
INFO: 13 processes: 13 internal.
FAILED: Build did NOT complete successfully
ERROR: Build failed. Not running target
^C
[1]+  Exit 1                  bazel run //tools:meldis -- -w
```
When I run it again, I see the following
```
bazel run //tools:meldis -- -w &
[1] 98883
INFO: Analyzed target //tools:meldis (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
ERROR: /home/murtaza/drake/bindings/pydrake/solvers/BUILD.bazel:41:21: Compiling bindings/pydrake/solvers/solvers_py_mathematicalprogram.cc failed: (Exit 1): gcc failed: error executing command (from target //bindings/pydrake/solvers:__init__.cpython-310-x86_64-linux-gnu.so) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 231 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
gcc: fatal error: Killed signal terminated program cc1plus
compilation terminated.
Target //bindings/pydrake/visualization:meldis failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 39.810s, Critical Path: 39.08s
INFO: 14 processes: 13 internal, 1 linux-sandbox.
FAILED: Build did NOT complete successfully
ERROR: Build failed. Not running target
```

jwnimmer-tri, 2023-03-13:
See https://drake.mit.edu/troubleshooting.html#build-oom.

ggould-tri, 2023-03-13:
Assigning @jwnimmer-tri as component lead.

AarishShah22, 2024-04-13:
I faced a similar issue. I am working on WSL with Ubuntu 22.04.3. I tried the fixes mentioned in the link above ( https://drake.mit.edu/troubleshooting.html#build-oom), but that didn't work. I also tried `bazel build --jobs=1 //tools:meldis` in case my device just didn't have enough resources, but that still gives me the same error message:

```
INFO: Analyzed target //tools:meldis (0 packages loaded, 0 targets configured).
ERROR: /home/aarishshah/drake_ws/drake/bindings/pydrake/multibody/BUILD.bazel:125:21: Compiling bindings/pydrake/multibody/plant_py.cc failed: (Exit 1): gcc failed: error executing CppCompile command (from target //bindings/pydrake/multibody:plant.cpython-310-x86_64-linux-gnu.so) /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 139 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
gcc: fatal error: Killed signal terminated program cc1plus
compilation terminated.
Target //bindings/pydrake/visualization:meldis failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 70.780s, Critical Path: 67.39s
INFO: 10 processes: 10 internal.
ERROR: Build did NOT complete successfully
```

The error message seems to match what the link describes, but the fix isn't working. Any help would be appreciated, thank you!