RobotLocomotion/drake #18982: CollisionCheckerParams option to disable preallocated contexts

jwnimmer-tri, 2023-03-13:
The `CollisionChecker::AllocateContexts()` function (called from the constructor) pre-allocates collision checking contexts so that the checker can be used in an OpenMP par-for loop.  For example, my computer has 48 vcpus, so the class constructor 48 contexts.

In some cases, the user knows that they will not be using OpenMP calls into the checker, in which case these extra contexts are a waste of resources.

We should consider adding a flag to `CollisionCheckerParams` for the user to opt-out of the per-thread contexts.  If they opt-out but still try to use the checker in a par-for loop, the checker should throw an exception.

\CC @calderpg-tri FYI.

jwnimmer-tri, 2023-04-10:
As a work-around for now, you can use something like this class to disable OpenMP prior to creating a CollisionChecker:

https://github.com/ToyotaResearchInstitute/common_robotics_utilities/blob/89435ebf4621f4e92d3b4c2a18aec7fa91e0cb75/include/common_robotics_utilities/openmp_helpers.hpp#L181

Any future clones of that checker will also be throttled to just use 1 thread & 1 context.

Setting `OMP_NUM_THREADS=1` environment variable is probably another option.

jwnimmer-tri, 2023-08-30:
Maybe a good answer here would be to add `Parallellism parallelism` to the `CollisionCheckerParams` struct?

@calderpg-tri WDYT?

calderpg-tri, 2023-08-30:
Yes, my thinking is to replace the current logic in `AllocateContexts` with a provided `Parallelism` (defaulted to `Parallelism::Max()`). The same holds for `PerThreadRandomSource` in anzu, which uses essentially identical logic for allocating per-thread generators.

PR #20138 (2023-09-07): [planning] CollisionChecker parallelism updates
Changes:

- Removes uses of CRU parallelization macros to decrease Drake+anzu+CRU coupling
- Provides `Parallelism` control to all parallel/optionally-parallelized `CollisionChecker` methods (many of which previously took just a `bool parallelize`)
- Provides an `implicit_context_parallelism` member of `CollisionCheckerParams` to optionally specify a different number of implicit contexts (e.g. to effectively disable implicit context parallelism if unnecessary), resolves #18982 
- Provides optional `context_number` parameters to all `CollisionChec [... cut]
commit 63ec447a (2023-09-07): [planning] CollisionChecker parallelism updates (#20138)

- Removes uses of CRU DegreeOfParallelism and parallelization macros to decrease Drake+anzu+CRU coupling
- Provides Parallelism control to all parallel/optionally-parallelized CollisionChecker methods (many of which previously took just a bool parallelize)
- Provides an implicit_context_parallelism member of CollisionCheckerParams to optionally specify a different number of implicit contexts (e.g. to effectively disable implicit context parallelism if unnecessary), resolves #18982
- Provides optional context_number parameters to all Col [... cut]
closed 2023-09-07 by calderpg-tri