RobotLocomotion/drake #18984: RollPitchYaw API suggestion for getting RpyDt from AngularVelocityInChild

youngiltrictr, 2023-03-13:
**Is your feature request related to a problem? Please describe.**
It's not a critical problem but just for completeness, I think it would be nice that `RollPitchYaw` provides `CalcRpyDtFromAngularVelocityInChild` too. It only provides `CalcRpyDtFromAngularVelocityInParent`.

For the other direction, `RollPitchYaw` provides two methods
`CalcAngularVelocityInChildFromRpyDt` and `CalcAngualrVelocityInParentFromRpyDt`

jwnimmer-tri, 2023-03-16:
FYI This issue arose from [a slack thread](https://app.slack.com/client/T0JNB2DS4/C3YB3EV5W/thread/C3YB3EV5W-1677807690.318249).

One suggestion there was:

> In the meantime, since  w_AD_A = R_AD * w_AD_D,  you can use something like:
CalcRpyDtFromAngularVelocityInParent( R_AD *  w_AD_D);
where w_AD_D is body D's angular velocity in A, expressed in D, i.e., AngularVelocityInChild.

mitiguy, 2023-03-17:
PR #19018 helps address issue #18984

mitiguy, 2023-03-27:
PR [19074](https://github.com/RobotLocomotion/drake/pull/19074) resolves issue [18984](https://github.com/RobotLocomotion/drake/issues/18984).

closed 2023-04-10 by mitiguy