RobotLocomotion/drake #18988: Drake fails to build from source

edrumwri, 2023-03-13:
### What happened?

I did a binary install to `/usr` of Drake (commit hash cd0afb) on Ubuntu Jammy. When attempting to parse (`Parser(mbp, scene_graph).AddModelFromFile(model_file_path, model_instance_name)`, I get the error:

```
C++ exception with description "PackageMap::GetPackageName(): Failed to parse XML in file "/share/drake_models/package.xml".
XML_ERROR_FILE_NOT_FOUND" thrown in ...
```

jwnimmer-tri, 2023-03-13:
Which version of Ubuntu?

jwnimmer-tri, 2023-03-13:
I'm currently running the following to see if I can reproduce the problem:

```
jwnimmer@call-cps:~$ docker run -it ubuntu:22.04
cd root
mkdir src
cd src
apt update
apt install git
git clone --filter=blob:none https://github.com/RobotLocomotion/drake.git
cd drake
git checkout cd0afb
setup/ubuntu/install_prereqs.sh --without-clang --without-test-only
mkdir ../build
cd build
cmake ../drake -DCMAKE_INSTALL_PREFIX=/usr
make install
```

jwnimmer-tri, 2023-03-13:
@edrumwri I could not reproduce the problem (see above for my attempted steps).

Could you provide a recipe for how I can reproduce the issue?

jwnimmer-tri, 2023-03-13:
In particular, the error message `/share/drake_models/package.xml` has lost the `/usr` somehow.  Ubuntu uses a symlink `/lib -> /usr/lib`.  Possibly you linked against `/lib/libdrake.so` instead of `/usr/lib/libdrake.so`?

edrumwri, 2023-03-13:
Will try this on our CI machine which also uses Docker to see if it's just a function of multiple Drake installs, as you suggested. If not, I'll provide some minimal source to reproduce.

jwnimmer-tri, 2023-03-14:
For the record, in case resource finding is malfunctioning, you can set an environment variable to explicitly direct it where to look:

```
# Set such that $DRAKE_RESOURCE_ROOT/drake/.drake-find_resource-sentinel exists.
export DRAKE_RESOURCE_ROOT=/usr/share
```

edrumwri, 2023-03-15:
Got it. I'll report on this as soon as Joe's PR https://reviewable.io/reviews/RobotLocomotion/drake/18983 lands, at which time we'll pin our codebase to that version of Drake (I was bringing our codebase up to date with Drake when I reported this issue).

jwnimmer-tri, 2023-03-28:
Closing for lack of details, and because the code that generated the error message has been rewritten anyway now.

closed 2023-03-28 by jwnimmer-tri